SpikeInterface/spikeinterface #1089: [implementation] Implement "nearest channel" localization method

alejoe91, 2022-11-23:
### Current issue

The current [`peak_localization`](https://github.com/SpikeInterface/spikeinterface/blob/master/spikeinterface/sortingcomponents/peak_localization.py) module implements the `center_of_mass` and `monopolar_triangulation` methods.
Some applications might use a very simple approach that localizes a spike/unit based on the nearest channel location.

### Proposed solution

Implement an additional `nearest_channel` method for `peak_localization`.

### ETA since in progress

1-2 weeks

### Other requirements

- [ ] Documentation
- [ ] Testing

h-mayorquin, 2022-11-28:
Right now the peak_detection method returns the following:
```
('sample_ind', 'channel_ind', 'amplitude', 'segment_ind')
```
For every peak. Where the `channel_ind` is the index of the channel in which that peak was detected. 

I am confused about the meaning of "nearest_channel" here, if the peaks only happen in the channel, the peak assigned should be the location of the  "channel_id", isn't it? In what sense do we look for the "nearest" channel if the spikes is only detected in the channel? 

Most likely there is something that I am misundesrtanding.

yger, 2022-11-28:
Indeed, the information is already located in 'channel_ind', since this is the channel where the peak is detected. However, peak_localization is returning positions. So the goal here would simply be to take the positions of the channels, the channels indices from the peaks, and return the positions (in x,y) of the peaks. In that sense, the "position" of a peak is the one of the nearest_channel, i.e. the one where the spikes was assigned

h-mayorquin, 2022-11-28:
Thanks, the goal I understand. I guess my point is more that "nearest_channel" seems misleading. 

"largest amplitude channel" or something of the sort would make more sense wouldn't it? Is there precedence for this name in some previous methodology?

yger, 2022-11-28:
Agreed. Largest amplitude channel might be a better wording

PR #1114 (2022-11-29): Add `peak_channel` as a method to `localize_peaks` in SortingComponents
Snould close #1089 once this is done.
closed 2022-11-29 by alejoe91