SpikeInterface/spikeinterface #1093: Add RecordingExtractor for NeuroExplorer

CodyCBakerPhD, 2022-11-23:
Neo RawIO here: https://github.com/NeuralEnsemble/python-neo/blob/master/neo/rawio/neuroexplorerrawio.py

Would be great to have a `NeuroExplorerRecordingExtractor` as well

Examples files: https://gin.g-node.org/NeuralEnsemble/ephy_testing_data/src/master/neuroexplorer

h-mayorquin, 2023-09-06:
See the coments here:
https://github.com/SpikeInterface/spikeinterface/issues/1773#issuecomment-1707859676

zm711, 2023-11-03:
@CodyCBakerPhD, just want to bump @h-mayorquin's message. Heberto implemented a reader for NeuroExplorer. Is that enough for your issue or does [216](https://github.com/catalystneuro/neuroconv/issues/216) require the most recent format? I'm trying to do a little post-release clean-up.

h-mayorquin, 2023-11-04:
I will ask the @CodyCBakerPhD for the data and open a new issue. I will close this.

closed 2023-11-04 by h-mayorquin