SpikeInterface/spikeinterface #1094: 'recording' must be a RecordingExtractor

olesnits, 2022-11-23:
Hello, when trying to use spiketoolkit.preprocessing.resample function with the recording extractor, i get such an error:

```
---------------------------------------------------------------------------
AssertionError                            Traceback (most recent call last)
<ipython-input-64-d2470865efa4> in <module>
----> 1 data_r = spiketoolkit.preprocessing.resample(recording, resample_rate = 10000)
```

D:\7 term KPI\python\lib\site-packages\spiketoolkit\preprocessing\resample.py in resample(recording, resample_rate)
     91 
     92     '''
---> 93     return ResampleRecording(
     94         recording=recording,
     95         resample_rate=resample_rate

D:\7 term KPI\python\lib\site-packages\spiketoolkit\preprocessing\resample.py in __init__(self, recording, resample_rate)
     21         assert HAVE_RR, "To use the ResampleRecording, install scipy: \n\n pip install scipy\n\n"
     22         self._resample_rate = resample_rate
---> 23         BasePreprocessorRecordingExtractor.__init__(self, recording, copy_times=False)
     24         self._dtype = recording.get_dtype()
     25 

D:\7 term KPI\python\lib\site-packages\spiketoolkit\preprocessing\basepreprocessorrecording.py in __init__(self, recording, copy_times)
      8 
      9     def __init__(self, recording, copy_times=True):
---> 10         assert isinstance(recording, RecordingExtractor), "'recording' must be a RecordingExtractor"
     11         RecordingExtractor.__init__(self)
     12         self._recording = recording

AssertionError: 'recording' must be a RecordingExtractor
```

alejoe91, 2022-11-24:
What's the output of `print(recording)`?

Can you make sure you are using the latest SI version? `pip install -U spikeinterface`

olesnits, 2022-11-24:
Alright. 
So, when I print recording I get this:
`MCSH5RecordingExtractor: 252 channels - 1 segments - 25.0kHz - 240.200s
  file_path: privateinfo/.h5`

I have checked and updated the spike interface version to the last one:
`SpikeInterface version: 0.96.1`

Still get the same error with the resample function:
`AssertionError: 'recording' must be a RecordingExtractor`

alejoe91, 2022-11-24:
Can you paste the full error?

alejoe91, 2022-11-24:
Also could you share the file so I can take a look? A google drive link would be great!

olesnits, 2022-11-24:
Can't unfortunately share a data due to hospital privacy. 
here is full error:
`---------------------------------------------------------------------------
AssertionError                            Traceback (most recent call last)
<ipython-input-40-f0636d4209d9> in <module>
----> 1 data_r = spiketoolkit.preprocessing.resample(data, resample_rate = 10000)

D:\7 term KPI\python\lib\site-packages\spiketoolkit\preprocessing\resample.py in resample(recording, resample_rate)
     91 
     92     '''
---> 93     return ResampleRecording(
     94         recording=recording,
     95         resample_rate=resample_rate

D:\7 term KPI\python\lib\site-packages\spiketoolkit\preprocessing\resample.py in __init__(self, recording, resample_rate)
     21         assert HAVE_RR, "To use the ResampleRecording, install scipy: \n\n pip install scipy\n\n"
     22         self._resample_rate = resample_rate
---> 23         BasePreprocessorRecordingExtractor.__init__(self, recording, copy_times=False)
     24         self._dtype = recording.get_dtype()
     25 

D:\7 term KPI\python\lib\site-packages\spiketoolkit\preprocessing\basepreprocessorrecording.py in __init__(self, recording, copy_times)
      8 
      9     def __init__(self, recording, copy_times=True):
---> 10         assert isinstance(recording, RecordingExtractor), "'recording' must be a RecordingExtractor"
     11         RecordingExtractor.__init__(self)
     12         self._recording = recording

AssertionError: 'recording' must be a RecordingExtractor`

alejoe91, 2022-11-24:
Oh you are using `spiketoolkit`! You should use the new API directly from spikeinterface! You can read about the version compatibility at the end of this page: https://spikeinterface.readthedocs.io/en/latest/

The "old" API is not maintained anymore..

alejoe91, 2022-11-24:
Actually, I think you are mixing the 2 APIs. Instead of `spiketoolkit.preprocessing.resample`, you should do:
```
import spikeinterface.preprocessing as spre

rec_r = spre.resample(recording, ...) 
```

olesnits, 2022-11-24:
oh yeah i just found this one:

olesnits, 2022-11-24:
'resample': <class 'spikeinterface.preprocessing.resample.ResampleRecording'>,

alejoe91, 2022-11-24:
Uninstall `spiketoolkit`!!! ;)

olesnits, 2022-11-24:
alright, it is working now. i just installed the spiketoolkit  to use the **resample** function. apparently it exists in the new API, you should mention this on the website for other people :)
Thank you!

closed 2022-11-24 by alejoe91