SpikeInterface/spikeinterface #1095: Spikeinterface not recognized in jupyter notebook

ABehal2020, 2022-11-24:
When I try to run the first block of code from the October 2022 spikeinterface tutorial jupyter notebook (https://github.com/SpikeInterface/spiketutorials/tree/master/Official_Tutorial_SI_0.96_Oct22), I get an error:

```
import spikeinterface as si
import spikeinterface.extractors as se 
import spikeinterface.preprocessing as spre
import spikeinterface.sorters as ss
import spikeinterface.postprocessing as spost
import spikeinterface.qualitymetrics as sqm
import spikeinterface.comparison as sc
import spikeinterface.exporters as sexp
import spikeinterface.widgets as sw
```

---------------------------------------------------------------------------
ModuleNotFoundError                       Traceback (most recent call last)
<ipython-input-1-a7cf23aa06e5> in <module>
----> 1 import spikeinterface as si
      2 import spikeinterface.extractors as se
      3 import spikeinterface.preprocessing as spre
      4 import spikeinterface.sorters as ss
      5 import spikeinterface.postprocessing as spost

ModuleNotFoundError: No module named 'spikeinterface'

I followed the instructions prior to launching the jupyter notebook by running `conda env create -f environment.yml`
I activated the created conda environment: `conda activate si_env`
I then added the conda environment to my jupyter notebook with the commands below:
```
conda install -c anaconda ipykernel
conda env create -f environment.yml
```
When I launched jupyter notebook, I made sure si_env was the kernel activated prior to trying to import spikeinterface.

ABehal2020, 2022-11-24:
turns out it was sys.path issue of python vs. ipython - if anyone else runs into this issue, this worked for me: https://jonathansoma.com/course/foundations-2021/quick-fix-jupyter-module-not-found/

closed 2022-11-24 by ABehal2020

alejoe91, 2022-11-24:
Great!