SpikeInterface/spikeinterface #1100: [feat-core] remove all classes (except for base) from init

alejoe91, 2022-11-24:
Since we encourage a functional approach, I suggest we remove all classes (e.g. for preprocessing, sorters, etc.) from the `__init__` file. This will make package exploration with tabs easier

@samuelgarcia?

samuelgarcia, 2022-11-24:
No sure.
For notebook, 
`rec, sorting = si.read_mearec()` si the way to go for me.
but in some pipeline having `rec = si.MEArecRecording` is also cool.
I think this could break lot of code for few benefit.
Lets think about this a bot more.

alejoe91, 2022-11-24:
why? I think the functional apporach is always easier to understand and to write (except for very advanced use cases, like `CollisionGTComparison` etc.)

alejoe91, 2022-11-24:
I haven't used a class in ages since the new API...

h-mayorquin, 2023-07-21:
Something to add to this debate is that the current form hampers tab discoverability by duplicating names:

[image]

I personally prefer classess but I would rather expose only one way of doing things.

zm711, 2024-06-19:
This is old. Should we close just because the default is to do nothing? (Though I agree with @alejoe91 for this one!)

h-mayorquin, 2024-06-19:
I also changed my mind to @alejoe91 position.  I personally use the classess but I also tend to do the full import path so this does not affect me and it will reduce the number  I wish there was a way of getting directly from the lower case version to the class directly though.

The concern of breaking code is valid though...

zm711, 2025-04-25:
So I've been looking into this a bit more and it seems a little more involved. Many of our inits use the class attributes so we import the class to get those attributes (like names etc). So let's bring this back up at the next maintenance meeting since Sam gave his okay we can briefly talk strategy for this.

alejoe91, 2025-06-06:
fixed by #3898

closed 2025-06-06 by alejoe91