SpikeInterface/spikeinterface #1107: Containerized Sorters problem

somalcy, 2022-11-25:
I tried to use example code from the link https://spikeinterface.readthedocs.io/en/latest/containerized_sorters.html#running-sorters-in-container-docker-singularity. It output the assertion error `must give a list of numpy array`

The environment is ubuntu 20.04, python 3.9, and spikeinterface 0.96.1.

The code for testing is
```
import spikeinterface.extractors as se
import spikeinterface.sorters as ss

test_recording, _ = se.toy_example(
    duration=30,
    seed=0,
    num_channels=64,
    num_segments=1
)

sorting = ss.run_waveclus(
    recording=test_recording,
    output_folder="waveclus",
    singularity_image=True)
```
 (I want to use waveclus sorter, and the kilosort3 sorter gives same assertion error)

The exception info:
```
SpikeSortingError: Spike sorting in singularity failed with the following error:
Traceback (most recent call last):
  File "/home/xxx/temp/in_container_sorter_script.py", line 9, in <module>
    recording = load_extractor('/home/cuilab/temp/in_container_recording.json')
  File "/home/xxx/temp/in_container_python_base/lib/python3.8/site-packages/spikeinterface/core/base.py", line 906, in load_extractor
    return BaseExtractor.load(file_or_folder_or_dict, base_folder=base_folder)
  File "/home/xxx/temp/in_container_python_base/lib/python3.8/site-packages/spikeinterface/core/base.py", line 545, in load
    extractor = BaseExtractor.from_dict(d, base_folder=base_folder)
  File "/home/xxx/temp/in_container_python_base/lib/python3.8/site-packages/spikeinterface/core/base.py", line 372, in from_dict
    extractor = _load_extractor_from_dict(d)
  File "/home/xxx/temp/in_container_python_base/lib/python3.8/site-packages/spikeinterface/core/base.py", line 849, in _load_extractor_from_dict
    extractor = cls(**kwargs)
  File "/home/xxx/temp/in_container_python_base/lib/python3.8/site-packages/spikeinterface/core/numpyextractors.py", line 30, in __init__
    assert all(isinstance(e, np.ndarray) for e in traces_list), 'must give a list of numpy array'
AssertionError: must give a list of numpy array
```
Why does this problem arise? waveclus sorter only uses CPU, so CUDA should not be the reason.

somalcy, 2022-11-26:
The code in `in_container_sorter_script.py` leads this problem. The `BaseRecording` object is saved to the storage via json. But it use `numpyextractors` to read the object, and json module can't convert the data to `numpy`, then raise an exception.

alejoe91, 2022-11-26:
Exactly, for many operations you need an object that can be read from disk, which is no the case for numpy. You can save the recording to fix it:
`test_recording = test_recording.save()`

alejoe91, 2022-11-28:
@somalcy is this solved?

somalcy, 2022-11-28:
> 

I tried `save` method and it convert the NumpyRecording to BinaryFolderRecording. I think it might work. I tried NwbRecording and the problem is solved. So others should work well except NumpyRecording. The example code in tutorial needs to be fixed.

alejoe91, 2022-11-28:
Thanks for the report! I'll fix the example tomorrow

somalcy, 2022-11-29:
BTW, is there any way to use multi-process in waveclus container? I checked it in `htop` and found it only use one core during sorting. @alejoe91

alejoe91, 2022-11-29:
Not sure! @ferchaure ??

PR #1115 (2022-11-29): [doc] Update container docs with save() function
Closes #1107 by extending documentation
closed 2022-11-29 by samuelgarcia