SpikeInterface/spikeinterface #1121: Potential bug in detect_peaks

hornauerp, 2022-11-30:
I tried to sort a MaxWell chip recording (10min, ~9gb) that I had previously sorted with an older spykingcircus version, using the spikeinterface-based sorters (spykingcircus 2 and tridesclous 2) with default parameters - other than `sorter_params['job_kwargs']['n_jobs'] = 8`. However, while there are clear spikes in the recording, both sorters did not find any peaks and gave an error afterwards. I went line by line through the spykingcircus2 code, and right before the `detect_peaks `call, there are still clear spikes in the data (example of one channel):
[image]

Interestingly, when I used `method=by_channel`, a few peaks were found, but they all belonged to the same channel. When I lowered the detection threshold to `sorter_params['detection']['detect_threshold'] = 4,` the number of peaks increased quite a bit, but still all peaks belonged to the same channel and the amplitudes also were very odd:
[image]

When running tridesclous2, I additionally get the following warnings and also detect 0 peaks:
```
E:\envs\si\lib\site-packages\spikeinterface\preprocessing\normalize_scale.py:230: RuntimeWarning: divide by zero encountered in true_divide
  gain = 1 / mads
E:\envs\si\lib\site-packages\spikeinterface\preprocessing\normalize_scale.py:231: RuntimeWarning: invalid value encountered in true_divide
  offset = -medians / mads
```

Since the sorting worked with the old spykingcircus, and the preprocessing does not seem to distort the data (clear spikes visible), there might be a bug in the `detect_peaks `function.

yger, 2022-11-30:
Thanks for the feedback. The bug is not necessarily in the detect_peak function. It could be in the way data are loaded/filtered, and or in the preprocessing steps. Could it be possible to share the data, or are they too big?

samuelgarcia, 2022-11-30:
Hi,
`detect_peaks()` have an important kwargs which is noise_level.
By default it is computed internally.
I would advice to compute outside with `get_noise_levels(rec)`  , so you can  check that this level is OK.

What is your preprocessing ?

yger, 2022-11-30:
Indeed, look at the distribution of noise levels, and maybe share the plot here. Just also, as a sanity check, maybe plotting some traces with si.plot_timeseries of the recording. Are you giving a raw recording to spyking circus 2 (and thus let the pipeline do the preprocessing), or a preprocessed recording (and with apply_preprocessing set to False in sc2)?

hornauerp, 2022-11-30:
Thanks for your quick replies! 

Initially, I did the preprocessing outside of the sorter (apply_preprocessing set to False), but to rule out any errors in my preprocessing pipeline, I now use the raw recording (so just `rec = si.MaxwellRecordingExtractor(rec_path)`) and set apply_preprocessing to True. However, both give the same result. 
I also checked `si.remove_bad_channels(rec)`, but it does not detect any bad channels. 

The histogram of the noise levels looks like this:
[image]

And I uploaded the [raw file ](https://polybox.ethz.ch/index.php/s/tk3dKoysodI882T) here if you have time to take a closer look.

yger, 2022-11-30:
i'll download the file and have to look to understand what is happening

yger, 2022-11-30:
I ran the following code

`
import spikeinterface.full as si
rec = si.read_maxwell('recording.raw.h5')
sub_rec = rec.frame_slice(0, 20000*60) ## To take only
from spikeinterface.sortingcomponents.peak_detection import detect_peaks
sub_rec = si.bandpass_filter(sub_rec, dtype='float32')
sub_rec = si.common_reference(sub_rec)
sub_rec = si.zscore(sub_rec)
peaks = detect_peaks(sub_rec, noise_levels=np.ones(1020), method='by_channel', n_jobs=-1, verbose=True, progress_bar=True, chunk_memory="100M")
import pylab as plt
plt.hist(peaks['channel_ind'], 100)
`
[image]

yger, 2022-11-30:
Actually, same with locally_exclusive, so except if the bug is after the first 60s, evertyhing looks fine to me. Could you double test on your side with the latest git version, and see if you have similar results? The zscore function is a default in SC2, such that all channels have the same noise levels of 1

hornauerp, 2022-12-01:
I pulled the latest master branch from github and ran the exact same code (except for explicitly assigning n_jobs ( #867 )), but I still get the same result:
[image]

yger, 2022-12-01:
Hum... So you ran the code I wrote in the cell, and we have a major discrepancy... Damned. 
Can you share the details of your env, such that I can see if I could reproduce it? This was performed with python 3.8

yger, 2022-12-01:
The default code was ran on windows, or on linux?

hornauerp, 2022-12-01:
I initially worked locally on windows.

alejoe91, 2023-06-12:
@hornauerp any update on this?

hornauerp, 2023-06-13:
I have resorted to work with KS again, so unfortunately no news from my side.

alejoe91, 2023-11-06:
@hornauerp ok to close this?

hornauerp, 2023-11-06:
@alejoe91 Yes, I will close it 👍

hornauerp, 2023-11-06:
Yes, I will close it 👍