SpikeInterface/spikeinterface #1122: problems with trying to get spikeinterface to work

LaurenzMuessig, 2022-11-30:
Hi,

I have set up spikeinterface in a virtual environment on a machine running Ubuntu and I am trying to run the docker image of kilosort3. The docker setup and nvidia integration generally seems to work (I can successfully run the test that nvidia suggests). 
I am trying to run this toy example:

import spikeinterface.extractors as se
import spikeinterface.sorters as ss
test_recording, _ = se.toy_example(
    duration=30,
    seed=0,
    num_channels=64,
    num_segments=1
)
sorting = ss.run_kilosort3(
    recording=test_recording,
    output_folder="kilosort3",
    singularity_image=True)
print(sorting)   

and I get this error in the call to kilosort:
---------------------------------------------------------------------------
SpikeSortingError                         Traceback (most recent call last)
Cell In [3], line 1
----> 1 sorting = ss.run_kilosort3(
      2     recording=test_recording,
      3     output_folder="/data/laurenz/",
      4     docker_image=True)
      6 print(sorting)

File ~/spikeInterface/lib/python3.8/site-packages/spikeinterface/sorters/runsorter.py:691, in run_kilosort3(*args, **kwargs)
    690 def run_kilosort3(*args, **kwargs):
--> 691     return run_sorter('kilosort3', *args, **kwargs)

File ~/spikeInterface/lib/python3.8/site-packages/spikeinterface/sorters/runsorter.py:137, in run_sorter(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, docker_image, singularity_image, with_output, **sorter_params)
    135         else:
    136             container_image = singularity_image
--> 137     return run_sorter_container(
    138         container_image=container_image,
    139         mode=mode,
    140         **common_kwargs,
    141     )
    143 return run_sorter_local(**common_kwargs)

File ~/spikeInterface/lib/python3.8/site-packages/spikeinterface/sorters/runsorter.py:578, in run_sorter_container(sorter_name, recording, mode, container_image, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, with_output, extra_requirements, **sorter_params)
    576 if run_error:
    577     if raise_error:
--> 578         raise SpikeSortingError(
    579             f"Spike sorting in {mode} failed with the following error:\n{run_sorter_output}")
    580 else:
    581     if with_output:

SpikeSortingError: Spike sorting in docker failed with the following error:
b'Traceback (most recent call last):\n  File "/data/in_container_sorter_script.py", line 9, in <module>\n    recording = load_extractor(\'/data/in_container_recording.json\')\n  File "/home/miniconda3/lib/python3.8/site-packages/spikeinterface/core/base.py", line 906, in load_extractor\n    return BaseExtractor.load(file_or_folder_or_dict, base_folder=base_folder)\n  File "/home/miniconda3/lib/python3.8/site-packages/spikeinterface/core/base.py", line 545, in load\n    extractor = BaseExtractor.from_dict(d, base_folder=base_folder)\n  File "/home/miniconda3/lib/python3.8/site-packages/spikeinterface/core/base.py", line 372, in from_dict\n    extractor = _load_extractor_from_dict(d)\n  File "/home/miniconda3/lib/python3.8/site-packages/spikeinterface/core/base.py", line 849, in _load_extractor_from_dict\n    extractor = cls(**kwargs)\n  File "/home/miniconda3/lib/python3.8/site-packages/spikeinterface/core/numpyextractors.py", line 30, in __init__\n    assert all(isinstance(e, np.ndarray) for e in traces_list), \'must give a list of numpy array\'\nAssertionError: must give a list of numpy array\n'

Any chance someone could help me out and point me to what the problem is? Thanks for your help.

Cheers

Laurenz

alejoe91, 2022-11-30:
Hi @LaurenzMuessig 

Check out this issue (same problem, using docker instead): https://github.com/SpikeInterface/spikeinterface/issues/1107

You need to `save()` the toy example before. I have updated the documentation as well:
https://spikeinterface.readthedocs.io/en/latest/containerized_sorters.html

LaurenzMuessig, 2022-12-01:
Hey @alejoe91. Thanks for your quick reply. Awesome, that fixed the problem and it now seems to be running fine. I'll test it on some actual data now. 
Cheers
Laurenz

closed 2022-12-01 by LaurenzMuessig

alejoe91, 2022-12-01:
Great! Good luck :)