SpikeInterface/spikeinterface #1123: [implementation] tPCA component

alejoe91, 2022-11-30:
### Current issue

PCA is a common way to reduce waveform dimensionality and extract features. The Paninski lab uses a temporal PCA (tPCA) approach, in which waveforms are first extracted on a subset of the recording (on a sabset of channels close to the peak), they are stacked (from `(num_spikes, num_samples, num_channels)` to `(num_spikes * num_channels, num_samples)`), and a PCA model is trained (the same model is applied to all channels). The same model is then used to reduce dimensionality to `num_pca`, so that input waveforms are transformed from `(num_spikes, num_samples, num_channels)` to `(num_spikes, num_pca, num_channels )` (with `num_pca=8`). 

In spikeinterface, we do have the same mechanism in `postprocessing.compute_principal_components(..., method="by_channel_global")`, but not as a sorting component.

### Proposed solution

Implement a `WaveformTemporalPCA` class which inherits from `PeakPipelineStep`.

The class should have a `fit()` method that looks as follows"
```
def fit(self, recording, **kwargs)
```
This function, internally, should select a portion of the recording (maybe random chunks), detect peaks and extract waveforms, compute the PCA model, and save it to `pkl` (it needs to be saved to allow for parellelization)

The `compute_buffer()` function, will apply the `model.transform()` function to the extracted waveforms in the buffer an return `tpcs_waveforms` with dimensions `(num_spikes, num_pca, num_channels )`

### ETA since in progress

2-3 weeks

### Other requirements

- [ ] Documentation
- [ ] Testing

samuelgarcia, 2023-01-17:
Hi @h-mayorquin.
I am sorry to make feedback with so much delay.
I did some comments dirtectly in the code.

h-mayorquin, 2023-03-06:
I am closing this as we have merged https://github.com/SpikeInterface/spikeinterface/pull/1245

closed 2023-03-06 by h-mayorquin