SpikeInterface/spikeinterface #1124: [infrastructure] Make `BaseWaveformTransformer` class

alejoe91, 2022-11-30:
### Current issue

Currently, the [`PeakPipeline`](https://github.com/SpikeInterface/spikeinterface/blob/master/spikeinterface/sortingcomponents/peak_pipeline.py) mechanism uses enables to detect peaks, extract waveforms on-the-fly, and do a series of computations on the waveforms (with `PeakPipelineSteps`).

However, extracted waveforms could be further processed before making the computations (e.g. cleaning)

### Proposed solution

We propose to make a `BaseWaveformTransformer` class with a `transform()` method (and optionally a `fit()`) method. During peak detection, a list of `WaveformTransformer` instances can be passed to the `detect_peaks()` function, so that extracted waveforms are processed on-the-fly before further processing.

The `WaveformDenoiser` will inherit from this base class.

### ETA since in progress

2-3 weeks

### Other requirements

- [ ] Documentation
- [ ] Testing

samuelgarcia, 2023-01-11:
Note that we need something a bit more flexible because what we also need is to be able to clean the waveform inside a peak detection method itself because we want to itertivelyy detect on the residual after a "clean".

alejoe91, 2023-01-11:
@samuelgarcia as we discussed the peak detection method could optionally return waveforms, no? So the iterative subtraction + cleaning will be a specific PeakDetection method

samuelgarcia, 2023-01-11:
Yes it is what I mean this futur brand new peak detection method will also need this waveform cleaner. So it will no be used only as a step in the chain but also (and mainly) inside a method. So a functional prroach will be certainly be needed.

h-mayorquin, 2023-02-10:
@alejoe91 @samuelgarcia should we close this one?

closed 2023-02-10 by samuelgarcia