SpikeInterface/spikeinterface #1125: phase_shift preprocessing forces recording.save( ) to write in float32 ?

maxjuv, 2022-11-30:
Hello,
I used the bandpass, phase_shift and common_ref preprocessing on my Neuropixel data. Then I save the recording. It appears my preprocessed recording is saved in float32 ('<f4' according to the binary.json), while the raw data is int16. 
I looked in the code, bandpass and commen-ref do not alter recording dtype. However, phase_shift changes the recording dtype to be float32. So, if dtype is not specified in the recording.save(), the recording size will doubled. The new argument 'force_dtype_back' is not used by the recording.save function. Is it on purpose ?
Thank you

samuelgarcia, 2022-11-30:
SAlut MAxime,
It is a bug.
Strange because `if self.force_dtype_back:` you force back int.

What is preprocess order ?

maxjuv, 2022-11-30:
Options 1, 2 and 3 return recording.get_type() = int16 and recording.save() writes in int16 (<i2).
As soon as I add the phase_shift in the preprocessing, recording.get_type() returns float32 and recording.save writes in float32 as well. 
I am wondering if the issue comes from the two following lines. L55 (https://github.com/SpikeInterface/spikeinterface/blob/f0692dd3e128788bc1dbee7d5cd5f36365ebc9ac/spikeinterface/preprocessing/phase_shift.py#L55) dtype is defined as float32 for computation. And at that line 75 (https://github.com/SpikeInterface/spikeinterface/blob/f0692dd3e128788bc1dbee7d5cd5f36365ebc9ac/spikeinterface/preprocessing/phase_shift.py#L75), it seems the recording dtype is definitively changes for float32 ?

alejoe91, 2022-12-28:
Found the issue and pushed a fix here #1172

closed 2023-01-26 by alejoe91