SpikeInterface/spikeinterface #1126: load_extractor('spikeinterface_recording.json') fails for CorrectMotionRecoding

TomBugnon, 2022-11-30:
Hi @samuelgarcia 

I get this error at the end of `run_sorter` after applying motion correction to the recording.
(Kilosort finished properly, but I'd like to be able to load the recording actually used for sorting to run the waveform extractor)
Best

```
Traceback (most recent call last):
  File "/home/tbugnon/projects/shared_sortings/shared_sortings/scripts/run_sorting_pipeline.py", line 98, in <module>
    sorting_pipeline.run_pipeline()
  File "/home/tbugnon/projects/shared_sortings/ecephys/ecephys/wne/sglx/pipeline/sorting_pipeline.py", line 241, in run_pipeline
    self.run_sorting()
  File "/home/tbugnon/projects/shared_sortings/ecephys/ecephys/wne/sglx/pipeline/sorting_pipeline.py", line 215, in run_sorting
    ss.run_sorter(
  File "/home/tbugnon/projects/shared_sortings/spikeinterface/spikeinterface/sorters/runsorter.py", line 143, in run_sorter
    return run_sorter_local(**common_kwargs)
  File "/home/tbugnon/projects/shared_sortings/spikeinterface/spikeinterface/sorters/runsorter.py", line 165, in run_sorter_local
    sorting = SorterClass.get_result_from_folder(output_folder)
  File "/home/tbugnon/projects/shared_sortings/spikeinterface/spikeinterface/sorters/basesorter.py", line 283, in get_result_from_folder
    recording = load_extractor(output_folder / 'spikeinterface_recording.json')
  File "/home/tbugnon/projects/shared_sortings/spikeinterface/spikeinterface/core/base.py", line 921, in load_extractor
    return BaseExtractor.load(file_or_folder_or_dict, base_folder=base_folder)
  File "/home/tbugnon/projects/shared_sortings/spikeinterface/spikeinterface/core/base.py", line 546, in load
    extractor = BaseExtractor.from_dict(d, base_folder=base_folder)
  File "/home/tbugnon/projects/shared_sortings/spikeinterface/spikeinterface/core/base.py", line 373, in from_dict
    extractor = _load_extractor_from_dict(d)
  File "/home/tbugnon/projects/shared_sortings/spikeinterface/spikeinterface/core/base.py", line 864, in _load_extractor_from_dict
    extractor = cls(**kwargs)
  File "/home/tbugnon/projects/shared_sortings/spikeinterface/spikeinterface/sortingcomponents/motion_correction.py", line 258, in __init__
    f = scipy.interpolate.interp1d(spatial_bins, operator(motion[:, :], axis=0), kind='linear',
TypeError: list indices must be integers or slices, not tuple

```

samuelgarcia, 2022-11-30:
At the moment the motion corrected is not dumpable.
I will try to add a ptach for that.
The only solution so is to save it in binary and read it back.

alejoe91, 2023-02-17:
@samuelgarcia this is fixed right?

samuelgarcia, 2023-02-17:
normaly yes.

closed 2023-02-17 by alejoe91