SpikeInterface/spikeinterface #1127: Kilosort3 parameters documentations

jonathanlin1210, 2022-11-30:
Hi, 

I am working on maxwell data with kilosort3 sorters and would like to play with the sorter parameters but couldn't find any documentation for what the parameters do. Is there anywhere I can find documentation/descriptions about what each parameter does?
Attached are the KS3 parameters that I am trying to work with.
[image]

Thanks,

samuelgarcia, 2022-12-01:
`Kilosort3Sorter.params_description()` should give you info.

The best is to directly ask kilosort devs.

Note that is you use maxwell and do not have to much drift or motion I would recommand to use kilosort2.

alejoe91, 2022-12-01:
Actually, we also have a specific function to get parameter descriptions:
```
import spikeinterface.sorters as ss
ss.get_sorter_params_description("kilosort3")
```

I agree with @samuelgarcia: try Kilosort2 first :)

closed 2022-12-02 by alejoe91