SpikeInterface/spikeinterface #1128: decentralised drift estimation and deepinterpolation

LaurenzMuessig, 2022-12-01:
Hi,

I was wondering if these are both already included in the spikeInterface pre-processing pipeline and if so how can I call them from the pre-processing module? Is there any documentation on this by any chance? Thanks for your help.  

https://proceedings.neurips.cc/paper/2021/file/b950ea26ca12daae142bd74dba4427c8-Paper.pdf
https://pubmed.ncbi.nlm.nih.gov/34650233/

Cheers

Laurenz

alejoe91, 2022-12-02:
Hi @LaurenzMuessig 

- The decentralized motion correction is integrated! We are still working on finalizing the module before adding full documentation, but maybe @samuelgarcia can show you how to run it (then you could switch KS motion correction off!)
- Deepinterpolation is also implemented, but it's very preliminary and you need a pre-trained model locally. If you have the model, you can use:
```
from spikeinterface.preprocessing import deepinterpolate

recording_di = deepinterpolate(recording, model_path)
```

Note that we currently have an ongoing project to: 
1. implement deepinterpolation training in SpikeInterface
2. share pre-trained deepinterpolation models

Stay tuned :)

samuelgarcia, 2022-12-02:
@LaurenzMuessig : for the motion_estimation/motion_correction: it is usabale now, it is in the master.
Doc is not done yet, but I can send a notebook if you want.

The API could change a bit in the next following weeks so do not implement pipeline that strongly depend on the actual master.

LaurenzMuessig, 2022-12-02:
HI @alejoe91 and @samuelgarcia,

Thank you so much for your quick responses. That all sounds great. It'd be very helpful, If you could share that notebook @samuelgarcia. I am more of a Matlab man and only started to get my head around python. So this would be very helpful. Cheers
Laurenz

LaurenzMuessig, 2022-12-08:
I have been trying to get the motion correction to work, and I have a question - what is the 'pipeline_steps' option expecting as argument to do the peak localisation in the peak pipeline when running 'detect_peaks'? Also can I run the motion estimation/corrrecttion as part of the peak pipeline as well?  Thanks for your help. 

Cheers

Laurenz

yger, 2022-12-08:
Using monopolar is slow compared to center_of_mass yes. But it is leading to better estimation of the positions, and thus of the drifts. The way to go for motion correction is to to 
detect_peaks()
select_peaks() # to keep only a subset of all the peaks, let's say 100000
localize_peaks()
Then you can estimate the motion for the subset of the peaks (and positions). Once the motion is estimated, you provide that to the CorrectMotionRecording object. Ultimately, this will be part of a single function, but for now all the steps are still exposed, because we are properly benchmarking everything

LaurenzMuessig, 2022-12-12:
Hi @yger,

Thanks for your message. Downsampling the peaks made it much more manageable thanks. 
Cheers
Laurenz

alejoe91, 2023-11-06:
Should be better explained here: https://spikeinterface.readthedocs.io/en/latest/how_to/handle_drift.html

closed 2023-11-06 by alejoe91