SpikeInterface/spikeinterface #1160: Using Waveform extractor on KS2.5 without recording file

musall, 2022-12-22:
Hi all,

I'm trying to use the waveform extractor to read kKlosort 2.5 output data from a Neuropixels recording to run the quality metrics module, However, for many of our recordings, we don't have the raw ap.bin file from SpikeGLX available.

Is it possible to use extract_waveforms() with 'with_recording=False' to achieve this?
I couldn't find documentation on this and was wondering how to use it without the .bin file and whether we could still compute all QMs in spikeinterface by using the kilosort templates.

Thanks a lot!
Simon

samuelgarcia, 2022-12-22:
Hi,
conceptualy, the WaveformExtractor is for extracting snippet of signals from the recording given spike times.
So the recording is really needed.

The `with_recording=False' is to **reload** a WaveformExtractor when the extraction is already done but the recording is not here anymore. It is a kind of "degraded" version of the WaveformExtractor for some plotting and computing that do not need anymore the signal.

By design, we extract only spike times for the sorters.
Everything else is recomputed.
So template, quality metrics, amplitudes are not extracted from any sorters.
They are all recomputed.

musall, 2022-12-22:
Hi Samuel,
That makes sense, thank you.

closed 2022-12-22 by musall