SpikeInterface/spikeinterface #1161: Unable to read neuralynx .ntt files

ABehal2020, 2022-12-22:
The block of code below yields an error when I'm trying to read neuralynx .ntt files.

`import spikeinterface.extractors as se 

folder_path_ntt = "/home/adityab/SpikeInterfaceTest/aditya-clustering/data"
recording_ntt = se.read_neuralynx(folder_path=folder_path_ntt)`

---------------------------------------------------------------------------
IndexError                                Traceback (most recent call last)
Cell In [1], line 4
      1 import spikeinterface.extractors as se 
      3 folder_path_ntt = "/home/adityab/SpikeInterfaceTest/aditya-clustering/data"
----> 4 recording_ntt = se.read_neuralynx(folder_path=folder_path_ntt)

File ~/spikeinterface/spikeinterface/core/core_tools.py:27, in define_function_from_class.<locals>.reader_func(*args, **kwargs)
     25 @copy_signature(source_class)
     26 def reader_func(*args, **kwargs):
---> 27     return source_class(*args, **kwargs)

File ~/spikeinterface/spikeinterface/extractors/neoextractors/neuralynx.py:29, in NeuralynxRecordingExtractor.__init__(self, folder_path, stream_id, stream_name, all_annotations)
     27 def __init__(self, folder_path, stream_id=None, stream_name=None, all_annotations=False):
     28     neo_kwargs = self.map_to_neo_kwargs(folder_path)
---> 29     NeoBaseRecordingExtractor.__init__(self, stream_id=stream_id, 
     30                                        stream_name=stream_name,
     31                                        all_annotations=all_annotations, 
     32                                        **neo_kwargs)
     33     self._kwargs.update(dict(folder_path=str(folder_path)))

File ~/spikeinterface/spikeinterface/extractors/neoextractors/neobaseextractor.py:62, in NeoBaseRecordingExtractor.__init__(self, stream_id, stream_name, block_index, all_annotations, **neo_kwargs)
     59         raise ValueError(f"This reader have several streams: \nNames: {stream_names}\nIDs: {stream_ids}. "
     60                          f"Specify it with the 'stram_name' or 'stream_id' arguments")
     61     else:
---> 62         stream_id = stream_ids[0]
     63         stream_name = stream_names[0]
     64 else:

IndexError: list index out of range

Here's a link to the zipped data folder with the ntt files: https://drive.google.com/file/d/1VzxCXIc7ZLL5_0kVqVULEYDVwGIlQWUf/view?usp=sharing

I would appreciate any help on this issue.

samuelgarcia, 2022-12-22:
Thanks for the report.
Could you please open an issue here : https://github.com/samuelgarcia/python-neo ?

alejoe91, 2022-12-22:
Hi, you are trying to read sorting data, so you should use the `read_neuralynx_sorting()` function. 
However, there is a bug on our side and I'm making a PR to fix it. Thanks for providing the test data!

alejoe91, 2022-12-22:
The function is called `read_neuralynx_sorting()` (it's [here](https://github.com/SpikeInterface/spikeinterface/blob/master/spikeinterface/extractors/neoextractors/neuralynx.py#L72)). Not that I found a bug, so you need to test from this PR: https://github.com/SpikeInterface/spikeinterface/pull/1166

closed 2022-12-22 by samuelgarcia

ABehal2020, 2022-12-22:
Given a spike interface extractor, a recording and sorting object may be generated. Does the recording object contain data that is ready to be preprocessed and subsequently spike sorted while the sorting object contains data already sorted for spike trains?

alejoe91, 2022-12-22:
Exactly! The .NTT contains sorted data, so you need the read sorting function

ABehal2020, 2023-04-03:
Yes

PR #1166 (2022-12-22): [bug-extractors] Fix neuralynx sorting function and NeoBaseSorting when no streams
fixes #1161 

In addition it fixes a case in `NeoBaseSortingExtractor` when sampling frequency cannot be auto-guessed