SpikeInterface/spikeinterface #1167: unable to load sorting object from `export_to_phy` output folder

a-zmz, 2022-12-22:
Hi,

After spike sorting, `export_to_phy` and manual curation, I tried to get template metrics of all units. However, I cannot load the latest sorting object into spikeinterface. Tried `se.read_kilosort` and `se.read_phy`, both gave me the same key error.
Have anyone encountered this? Cheers!
[image]

alejoe91, 2022-12-28:
@a-zmz the problem is that you have 2 `cluster_info` files. One is a csv (comma-separated), the other is a tsv (tab-separated).

The csv file is wrong. Despite the suffix, it's in fact a tab-separated file (and that's where the error stems from). You can just delete the `cluster_info.csv` file (it's the exact same file as `cluster_info.tsv`) and everything works as usual.

How did you get these duplicate files?

a-zmz, 2023-01-09:
Hi Alessio sorry for the late reply!

I got the `.csv` version by copying the `cluster_info.tsv` to accommodate my colleague's matlab code. I can confirm that by deleting the `.csv` the sorting object can be loaded correctly.

Thanks very much!

closed 2023-01-09 by a-zmz