SpikeInterface/spikeinterface #1168: Docker images return a strange error

rat-h, 2022-12-22:
I'm following the tutorial but want to run sorters in Docker container.
Call for any sorter with `docker_image=` causes the same kind of error.
For example

```python
sorting_TDC = ss.run_tridesclous(recording, docker_image="spikeinterface/tridesclous-base:latest")
```
```
AssertionError                            Traceback (most recent call last)
File ~/.local/apps/jupytor/lib/python3.10/site-packages/spikeinterface/sorters/runsorter.py:583, in run_sorter_container(sorter_name, recording, mode, container_image, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, with_output, extra_requirements, **sorter_params)
    582 try:
--> 583     sorting = SorterClass.get_result_from_folder(output_folder)
    584 except Exception as e:

File ~/.local/apps/jupytor/lib/python3.10/site-packages/spikeinterface/sorters/basesorter.py:281, in BaseSorter.get_result_from_folder(cls, output_folder)
    278     raise SpikeSortingError(
    279         "Spike sorting failed. You can inspect the runtime trace in spikeinterface_log.json")
--> 281 sorting = cls._get_result_from_folder(output_folder)
    283 recording = load_extractor(output_folder / 'spikeinterface_recording.json')

File ~/.local/apps/jupytor/lib/python3.10/site-packages/spikeinterface/sorters/tridesclous/tridesclous.py:160, in TridesclousSorter._get_result_from_folder(cls, output_folder)
    158 @classmethod
    159 def _get_result_from_folder(cls, output_folder):
--> 160     sorting = TridesclousSortingExtractor(folder_path=output_folder)
    161     return sorting

File ~/.local/apps/jupytor/lib/python3.10/site-packages/spikeinterface/extractors/tridesclousextractors.py:36, in TridesclousSortingExtractor.__init__(self, folder_path, chan_grp)
     35 def __init__(self, folder_path, chan_grp=None):
---> 36     assert self.installed, self.installation_mesg
     37     tdc_folder = Path(folder_path)

AssertionError: To use the TridesclousSortingExtractor install tridesclous: 

 pip install tridesclous

During handling of the above exception, another exception occurred:

ValueError                                Traceback (most recent call last)
Cell In[11], line 1
----> 1 sorting_TDC = ss.run_tridesclous(recording, docker_image="spikeinterface/tridesclous-base:latest")

File ~/.local/apps/jupytor/lib/python3.10/site-packages/spikeinterface/sorters/runsorter.py:649, in run_tridesclous(*args, **kwargs)
    648 def run_tridesclous(*args, **kwargs):
--> 649     return run_sorter('tridesclous', *args, **kwargs)

File ~/.local/apps/jupytor/lib/python3.10/site-packages/spikeinterface/sorters/runsorter.py:137, in run_sorter(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, docker_image, singularity_image, with_output, **sorter_params)
    135         else:
    136             container_image = singularity_image
--> 137     return run_sorter_container(
    138         container_image=container_image,
    139         mode=mode,
    140         **common_kwargs,
    141     )
    143 return run_sorter_local(**common_kwargs)

File ~/.local/apps/jupytor/lib/python3.10/site-packages/spikeinterface/sorters/runsorter.py:590, in run_sorter_container(sorter_name, recording, mode, container_image, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, with_output, extra_requirements, **sorter_params)
    586     print('Failed to get result with sorter specific extractor.\n'
    587           f'Error Message: {e}\n'
    588           'Getting result from in-container saved NpzSortingExtractor')
    589 try:
--> 590     sorting = NpzSortingExtractor.load_from_folder(npz_sorting_path)
    591 except FileNotFoundError:
    592     SpikeSortingError(f"Spike sorting in {mode} failed with the following error:\n{run_sorter_output}")

File ~/.local/apps/jupytor/lib/python3.10/site-packages/spikeinterface/core/base.py:576, in BaseExtractor.load_from_folder(folder)
    574 @staticmethod
    575 def load_from_folder(folder):
--> 576     return BaseExtractor.load(folder)

File ~/.local/apps/jupytor/lib/python3.10/site-packages/spikeinterface/core/base.py:572, in BaseExtractor.load(file_path, base_folder)
    569     return extractor
    571 else:
--> 572     raise ValueError('spikeinterface.Base.load() file_path must be an existing folder or file')

ValueError: spikeinterface.Base.load() file_path must be an existing folder or file

```

I can run this container manually
```bash
(jupytor) xxx@colonneselab-linux-workstation:/scratch/xxx/TEST_SpikeInterface$ docker image ls
REPOSITORY                               TAG                       IMAGE ID       CREATED        SIZE
nvidia/cuda                              11.3.0-base-ubi8          c226dc4dc637   5 days ago     256MB
spikeinterface/kilosort3-compiled-base   latest                    7bb20832614f   2 months ago   7.46GB
spikeinterface/kilosort-compiled-base    latest                    d3bffaad53cc   2 months ago   7.17GB
spikeinterface/tridesclous-base          latest                    a627d89f03dd   4 months ago   1.94GB
spikeinterface/yass-base                 latest                    9a10c120298c   4 months ago   7.72GB
nvidia/cuda                              11.0.3-base-ubuntu20.04   8017f5c31b74   5 months ago   122MB

(jupytor) xxx@colonneselab-linux-workstation:/scratch/xxx/TEST_SpikeInterface$ docker run -ti spikeinterface/tridesclous-base /bin/bash
root@095b809bf79e:/# python
Python 3.8.13 (default, May 11 2022, 08:23:48)  
[GCC 10.2.1 20210110] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import tridesclous
>>>  
root@095b809bf79e:/# exit
exit
(jupytor) xxx@colonneselab-linux-workstation:/scratch/xxx/TEST_SpikeInterface$
```

alejoe91, 2022-12-27:
Hi @rat-h 

What SpikeInterface version are you using? Is it master?

What is the content of the output folder? 

Thanks for the report!

rat-h, 2022-12-27:
Hi @alejoe91 
SpikeIterface version - I actually don't know. Installed by pip version 0.96.1 (I reinstalled everything since, so it is currently installed version) 
Output folder was empty.

A bit more details I forgot to report. I'm running spikeinterface in a virtual environment (as you can see from the prompt).

Since my bug report I found that docker runs once in a while like 1-2 times over 10 attempt. If container runs it generates correct output.

**Update**
I switched to singularity. After fresh install this problem is gone. This problem is obviously on the side of docker interface

alejoe91, 2022-12-27:
Happy that it works smoothly with singularity. I'll close the issue since it seems to be due to docker issues, rather than SpikeInterface itself

closed 2022-12-27 by alejoe91