SpikeInterface/spikeinterface #1169: Sorting Discrete Putative Spikes

ABehal2020, 2022-12-22:
I have some .ntt files I would like to spike sort (https://drive.google.com/file/d/1VzxCXIc7ZLL5_0kVqVULEYDVwGIlQWUf/view?usp=sharing). These files are a few years old, and the Neuralynx software collected the data in a discontinuous fashion to minimize the recording file size. These .ntt files do not contain raw 32 kHz traces but do contain unsorted sweeps of threshold-crossing moments (putative spikes). 

Is spike interface able to spike sort this discontinuous data with unsorted, putative spikes or only raw continuous traces?

alejoe91, 2022-12-22:
Only waveclus supports this kind of data, that we call snippets. However there is no automatic way to load it, so you would need to use `neo` to load the waveforms and then construct a `NumpySnippets` object, which you can then use with `waveclus_snippet`

ABehal2020, 2022-12-23:
This is what I have so far.

```
reader = nr.NeuralynxRawIO(dirname=folder_path_ntt)
reader.parse_header()

# Read spike waveforms - see https://neo.readthedocs.io/en/stable/examples.html
raw_waveforms = reader.get_spike_raw_waveforms()
float_waveforms = reader.rescale_waveforms_to_float(raw_waveforms)

sorter_params = ss.get_default_params('waveclus_snippets')
sorting_waveclus_snippets = ss.run_sorter('waveclus_snippets', float_waveforms,
                             output_folder=base_folder / 'results_waveclus_snippets',
                             verbose=True, docker_image=True, **sorter_params)
```

The type of float_waveforms is <class 'numpy.ndarray'>. 

However, this code block gives the following error: 

---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
Cell In [96], line 2
      1 sorter_params = ss.get_default_params('waveclus_snippets')
----> 2 sorting_waveclus_snippets = ss.run_sorter('waveclus_snippets', float_waveforms,
      3                              output_folder=base_folder / 'results_waveclus_snippets',
      4                              verbose=True, docker_image=True, **sorter_params)

File ~/spikeinterface/spikeinterface/sorters/runsorter.py:137, in run_sorter(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, docker_image, singularity_image, with_output, **sorter_params)
    135         else:
    136             container_image = singularity_image
--> 137     return run_sorter_container(
    138         container_image=container_image,
    139         mode=mode,
    140         **common_kwargs,
    141     )
    143 return run_sorter_local(**common_kwargs)

File ~/spikeinterface/spikeinterface/sorters/runsorter.py:381, in run_sorter_container(sorter_name, recording, mode, container_image, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, with_output, extra_requirements, **sorter_params)
    378 output_folder.mkdir(parents=True, exist_ok=True)
    380 # find input folder of recording for folder bind
--> 381 rec_dict = recording.to_dict()
    382 recording_input_folders = find_recording_folders(rec_dict)
    384 if platform.system() == 'Windows':

AttributeError: 'numpy.ndarray' object has no attribute 'to_dict'

How can I use the waveclus_snippets sorter?

alejoe91, 2022-12-27:
Hi @ABehal2020 

Almost there! You need to instantiate a `NumpySnippets` object from the extracted waveforms and spike frames (that you need to extract):

```
import spikeinterface as si

# retrieve spike frames (sample_indices for detected waveforms) from reader
spike_frames = [...]
sampling_frequency = 30000  # use correct frequency
nbefore = 30  # this is the number of samples before the peak
recording_snippets = si.NumpySnippets(float_waveforms, spike_frames, sampling_frequency, nbefore=nbefore)

# since this object is in-meory, we have to save it to a folder in order to use docker
recording_snipptes = recording_snippets.save(folder="some-folder")

# now you can run waveclus_snippets
...
sorting_waveclus_snippets = ss.run_sorter('waveclus_snippets', recording_snipptes,
                             output_folder=base_folder / 'results_waveclus_snippets',
                             verbose=True, docker_image=True, **sorter_params)
```

Note that while testing I found a small bug in saving snippets which is fixed here: https://github.com/SpikeInterface/spikeinterface/pull/1171

alejoe91, 2022-12-29:
@ABehal2020 as mentioned here https://github.com/SpikeInterface/spikeinterface/issues/1173#issuecomment-1367186811, you need an additional line to reshape your waveforms before calling the `NumpySnippets`.

I paste the code here:
```
folder_path_ntt = "/home/adityab/SpikeInterfaceTest/aditya-clustering/data"
reader = nr.NeuralynxRawIO(dirname=folder_path_ntt)
reader.parse_header()
# print(reader)

# Read spike waveforms - see https://neo.readthedocs.io/en/stable/examples.html
raw_waveforms = reader.get_spike_raw_waveforms()
float_waveforms = reader.rescale_waveforms_to_float(raw_waveforms)
float_waveforms = float_waveforms.swapaxes(1, 2)

# retrieve spike frames (sample_indices for detected waveforms) from reader
spike_frames = reader.get_spike_timestamps() 

### Use correct sampling_frequency and nbefore here
sampling_frequency = 10000                          
nbefore = 15                                        
recording_snippets = si.NumpySnippets(float_waveforms, spike_frames, sampling_frequency, nbefore=nbefore)

# since this object is in-meory, we have to save it to a folder in order to use docker
recording_snipptes = recording_snippets.save(folder="rec_snippets")

# run sorter
sorting_waveclus_snippets = si.run_sorter('waveclus_snippets', recording_snipptes,
                             output_folder='results_waveclus_snippets', docker_image=True,
                             verbose=True)
print(sorting_waveclus_snippets)
```

closed 2022-12-29 by alejoe91

ABehal2020, 2022-12-29:
Thank you! How can I find out the value of nbefore? Is it in the reader.header information somewhere?

alejoe91, 2022-12-31:
Not sure! You can also plot some waveforms and infer it from there. It's the same index that waveforms are aligned on ;)