SpikeInterface/spikeinterface #1170: Fix auto_merge randomness

alejoe91, 2022-12-23:
@samuelgarcia @DradeAW 

The `auto_merge()` function fails now and then: https://github.com/SpikeInterface/spikeinterface/actions/runs/3765257303/jobs/6400671903#step:23:102

Could you look into it?

DradeAW, 2022-12-23:
If I had to guess, I would say that the correlogram or template step is failing (few number of spikes means randomness.

https://github.com/SpikeInterface/spikeinterface/blob/fbc92d2c44f8954c62b15bc63765c44030f38ede/spikeinterface/curation/tests/test_auto_merge.py#L25

We could try setting a higher `duration` and/or `firing_rate` to see if things stabilize, but it might make the test take more time to run.

alejoe91, 2022-12-23:
@DradeAW my bad!! In a recent PR, I unified the return values of ISI_violations and RP_violations, but I didn't change it in the curation module. Fixed it now :) Sorry about the false blaming!
See commit https://github.com/SpikeInterface/spikeinterface/pull/1142/commits/46871fba4dfe801d4cdc55dc96bc2a311f19f320

closed 2022-12-23 by alejoe91