SpikeInterface/spikeinterface #1173: Sorting snippets from .nex file

mfvd, 2022-12-28:
Hello,
I'm trying to sort some data from a collaborator in which activity from cultured neurons was recorded on a MEA. Unfortunately I only have extracted spikes or snippets stored in a .nex file. I've tried the approach suggested in issue #1169 by Alessio but keep bumping into the error below.

Ant suggestion how to solve this?

```python
RUNNING SHELL SCRIPT: /var/folders/1q/x87fsmf571x62kz_721lvzdc0000gn/T/tmp_shellscriptue3fsx7f/script.sh
Num. channels = 31, Num. snippets = 22774
RUNNING SHELL SCRIPT: /var/folders/1q/x87fsmf571x62kz_721lvzdc0000gn/T/tmp_shellscript884dtq_g/script.sh
Running waveclus in /Volumes/DataProc/Sara/spike_sorting/results_waveclus_snippets...
RUNNING SHELL SCRIPT: /var/folders/1q/x87fsmf571x62kz_721lvzdc0000gn/T/tmp_shellscripteav1iitm/script.sh
RUNNING SHELL SCRIPT: /Volumes/DataProc/Sara/spike_sorting/results_waveclus_snippets/run_waveclus_snippets.sh
/Volumes/DataProc/Sara/spike_sorting/results_waveclus_snippets/run_waveclus_snippets.sh: line 3: matlab: command not found

Error running waveclus_snippets
Traceback (most recent call last):
  File "/opt/anaconda3/envs/si_openephys/lib/python3.8/site-packages/spikeinterface/sorters/basesorter.py", line 225, in run_from_folder
    SorterClass._run_from_folder(output_folder, sorter_params, verbose)
  File "/opt/anaconda3/envs/si_openephys/lib/python3.8/site-packages/spikeinterface/sorters/waveclus/waveclus_snippets.py", line 162, in _run_from_folder
    raise Exception('waveclus returned a non-zero exit code')
Exception: waveclus returned a non-zero exit code

Output exceeds the [size limit](command:workbench.action.openSettings?[). Open the full output data [in a text editor](command:workbench.action.openLargeOutput?858577cf-6fe8-4b6e-af38-df88d6b937e8)
---------------------------------------------------------------------------
SpikeSortingError                         Traceback (most recent call last)
/Volumes/DataProc/Sara/spike_sorting/MEA_sorting.ipynb Cell 7 in <cell line: 2>()
      [1](vscode-notebook-cell:/Volumes/DataProc/Sara/spike_sorting/MEA_sorting.ipynb#X10sZmlsZQ%3D%3D?line=0) # now you can run waveclus_snippets
----> [2](vscode-notebook-cell:/Volumes/DataProc/Sara/spike_sorting/MEA_sorting.ipynb#X10sZmlsZQ%3D%3D?line=1) sorting_waveclus_snippets = ss.run_sorter('waveclus_snippets', recording_snipptes,
      [3](vscode-notebook-cell:/Volumes/DataProc/Sara/spike_sorting/MEA_sorting.ipynb#X10sZmlsZQ%3D%3D?line=2)                              output_folder='results_waveclus_snippets',
      [4](vscode-notebook-cell:/Volumes/DataProc/Sara/spike_sorting/MEA_sorting.ipynb#X10sZmlsZQ%3D%3D?line=3)                              verbose=True, docker_image=False)

File /opt/anaconda3/envs/si_openephys/lib/python3.8/site-packages/spikeinterface/sorters/runsorter.py:142, in run_sorter(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, docker_image, singularity_image, with_output, **sorter_params)
    135             container_image = singularity_image
    136     return run_sorter_container(
    137         container_image=container_image,
    138         mode=mode,
    139         **common_kwargs,
    140     )
--> 142 return run_sorter_local(**common_kwargs)

File /opt/anaconda3/envs/si_openephys/lib/python3.8/site-packages/spikeinterface/sorters/runsorter.py:162, in run_sorter_local(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, with_output, **sorter_params)
    159 SorterClass.set_params_to_folder(
    160     recording, output_folder, sorter_params, verbose)
    161 SorterClass.setup_recording(recording, output_folder, verbose=verbose)
--> 162 SorterClass.run_from_folder(output_folder, raise_error, verbose)
    163 if with_output:
    164     sorting = SorterClass.get_result_from_folder(output_folder)
...
--> 261     raise SpikeSortingError(
    262         f"Spike sorting failed. You can inspect the runtime trace in {output_folder}/spikeinterface_log.json")
    264 return run_time

SpikeSortingError: Spike sorting failed. You can inspect the runtime trace in results_waveclus_snippets/spikeinterface_log.json
```

```
"name": "SpikeSortingError",
	"message": "Spike sorting failed. You can inspect the runtime trace in results_waveclus_snippets/spikeinterface_log.json",
	"stack": "\u001b[0;31m---------------------------------------------------------------------------\u001b[0m\n\u001b[0;31mSpikeSortingError\u001b[0m                         Traceback (most recent call last)\n\u001b[1;32m/Volumes/DataProc/Sara/spike_sorting/MEA_sorting.ipynb Cell 7\u001b[0m in \u001b[0;36m<cell line: 2>\u001b[0;34m()\u001b[0m\n\u001b[1;32m      <a href='vscode-notebook-cell:/Volumes/DataProc/Sara/spike_sorting/MEA_sorting.ipynb#X10sZmlsZQ%3D%3D?line=0'>1</a>\u001b[0m \u001b[39m# now you can run waveclus_snippets\u001b[39;00m\n\u001b[0;32m----> <a href='vscode-notebook-cell:/Volumes/DataProc/Sara/spike_sorting/MEA_sorting.ipynb#X10sZmlsZQ%3D%3D?line=1'>2</a>\u001b[0m sorting_waveclus_snippets \u001b[39m=\u001b[39m ss\u001b[39m.\u001b[39;49mrun_sorter(\u001b[39m'\u001b[39;49m\u001b[39mwaveclus_snippets\u001b[39;49m\u001b[39m'\u001b[39;49m, recording_snipptes,\n\u001b[1;32m      <a href='vscode-notebook-cell:/Volumes/DataProc/Sara/spike_sorting/MEA_sorting.ipynb#X10sZmlsZQ%3D%3D?line=2'>3</a>\u001b[0m                              output_folder\u001b[39m=\u001b[39;49m\u001b[39m'\u001b[39;49m\u001b[39mresults_waveclus_snippets\u001b[39;49m\u001b[39m'\u001b[39;49m,\n\u001b[1;32m      <a href='vscode-notebook-cell:/Volumes/DataProc/Sara/spike_sorting/MEA_sorting.ipynb#X10sZmlsZQ%3D%3D?line=3'>4</a>\u001b[0m                              verbose\u001b[39m=\u001b[39;49m\u001b[39mTrue\u001b[39;49;00m, docker_image\u001b[39m=\u001b[39;49m\u001b[39mFalse\u001b[39;49;00m)\n\nFile \u001b[0;32m/opt/anaconda3/envs/si_openephys/lib/python3.8/site-packages/spikeinterface/sorters/runsorter.py:142\u001b[0m, in \u001b[0;36mrun_sorter\u001b[0;34m(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, docker_image, singularity_image, with_output, **sorter_params)\u001b[0m\n\u001b[1;32m    135\u001b[0m             container_image \u001b[39m=\u001b[39m singularity_image\n\u001b[1;32m    136\u001b[0m     \u001b[39mreturn\u001b[39;00m run_sorter_container(\n\u001b[1;32m    137\u001b[0m         container_image\u001b[39m=\u001b[39mcontainer_image,\n\u001b[1;32m    138\u001b[0m         mode\u001b[39m=\u001b[39mmode,\n\u001b[1;32m    139\u001b[0m         \u001b[39m*\u001b[39m\u001b[39m*\u001b[39mcommon_kwargs,\n\u001b[1;32m    140\u001b[0m     )\n\u001b[0;32m--> 142\u001b[0m \u001b[39mreturn\u001b[39;00m run_sorter_local(\u001b[39m*\u001b[39;49m\u001b[39m*\u001b[39;49mcommon_kwargs)\n\nFile \u001b[0;32m/opt/anaconda3/envs/si_openephys/lib/python3.8/site-packages/spikeinterface/sorters/runsorter.py:162\u001b[0m, in \u001b[0;36mrun_sorter_local\u001b[0;34m(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, with_output, **sorter_params)\u001b[0m\n\u001b[1;32m    159\u001b[0m SorterClass\u001b[39m.\u001b[39mset_params_to_folder(\n\u001b[1;32m    160\u001b[0m     recording, output_folder, sorter_params, verbose)\n\u001b[1;32m    161\u001b[0m SorterClass\u001b[39m.\u001b[39msetup_recording(recording, output_folder, verbose\u001b[39m=\u001b[39mverbose)\n\u001b[0;32m--> 162\u001b[0m SorterClass\u001b[39m.\u001b[39;49mrun_from_folder(output_folder, raise_error, verbose)\n\u001b[1;32m    163\u001b[0m \u001b[39mif\u001b[39;00m with_output:\n\u001b[1;32m    164\u001b[0m     sorting \u001b[39m=\u001b[39m SorterClass\u001b[39m.\u001b[39mget_result_from_folder(output_folder)\n\nFile \u001b[0;32m/opt/anaconda3/envs/si_openephys/lib/python3.8/site-packages/spikeinterface/sorters/basesorter.py:261\u001b[0m, in \u001b[0;36mBaseSorter.run_from_folder\u001b[0;34m(cls, output_folder, raise_error, verbose)\u001b[0m\n\u001b[1;32m    259\u001b[0m \u001b[39mif\u001b[39;00m has_error \u001b[39mand\u001b[39;00m raise_error:\n\u001b[1;32m    260\u001b[0m     \u001b[39mprint\u001b[39m(log[\u001b[39m'\u001b[39m\u001b[39merror_trace\u001b[39m\u001b[39m'\u001b[39m])\n\u001b[0;32m--> 261\u001b[0m     \u001b[39mraise\u001b[39;00m SpikeSortingError(\n\u001b[1;32m    262\u001b[0m         \u001b[39mf\u001b[39m\u001b[39m\"\u001b[39m\u001b[39mSpike sorting failed. You can inspect the runtime trace in \u001b[39m\u001b[39m{\u001b[39;00moutput_folder\u001b[39m}\u001b[39;00m\u001b[39m/spikeinterface_log.json\u001b[39m\u001b[39m\"\u001b[39m)\n\u001b[1;32m    264\u001b[0m \u001b[39mreturn\u001b[39;00m run_time\n\n\u001b[0;31mSpikeSortingError\u001b[0m: Spike sorting failed. You can inspect the runtime trace in results_waveclus_snippets/spikeinterface_log.json"
}
```

alejoe91, 2022-12-28:
@ferchaure could you take a look at this? It seems indeed that waveclus is failing.

@mfvd since you're running the sorter locally, can you use the `verbose=True` option and paste the output here?

mfvd, 2022-12-28:
Thanks for the quick reply! The output above is with `verbose=True`.

ferchaure, 2022-12-28:
Hi @mfvd could you share with me the snippets and the parameters used?, I'm not sure but maybe is some border case not well handled.

mfvd, 2022-12-28:
Hi @ferchaure. You can find one example file [here](https://drive.google.com/file/d/1Vz0q2oDavmBQVaINMgZB1QeIT4g6zhsR/view?usp=sharing)

I've used the following code and left the sorter params as default.

```python
file = "/Volumes/DataProc/spike_sorting/mea1.nex"
reader = neo.io.NeuroExplorerIO(file)
reader.parse_header()
# print(reader)

# Read spike waveforms - see https://neo.readthedocs.io/en/stable/examples.html
raw_waveforms = reader.get_spike_raw_waveforms()
float_waveforms = reader.rescale_waveforms_to_float(raw_waveforms)

# retrieve spike frames (sample_indices for detected waveforms) from reader
spike_frames = reader.get_spike_timestamps()        ##### CONFIRM THIS!!! #####
sampling_frequency = 10000                          
nbefore = 15                                        
recording_snippets = si.NumpySnippets(float_waveforms, spike_frames, sampling_frequency, nbefore=nbefore)

# since this object is in-meory, we have to save it to a folder in order to use docker
recording_snipptes = recording_snippets.save(folder="rec_snippets")

# run sorter
sorting_waveclus_snippets = ss.run_sorter('waveclus_snippets', recording_snipptes,
                             output_folder='results_waveclus_snippets',
                             verbose=True)
```

alejoe91, 2022-12-29:
Hi guys,

I had a quick look and I think that the problem is the shape of the `float_waveforms` object.
The `NumpySnippets` expects snippets to have `(num_spikes, num_samples, num_channels)` shape, but in your case you have `(num_spikes, num_channels, num_samples)` (I assume is a single channel).

You can fix this by adding this line after defining `float_waveforms`:
```
float_waveforms = float_waveforms.swapaxes(1, 2)
```

Here's the full code:
```
file = "/home/alessio/Documents/data/debug/mea1.nex"
reader = neo.io.NeuroExplorerIO(file)
reader.parse_header()
# print(reader)

# Read spike waveforms - see https://neo.readthedocs.io/en/stable/examples.html
raw_waveforms = reader.get_spike_raw_waveforms()
float_waveforms = reader.rescale_waveforms_to_float(raw_waveforms)
float_waveforms = float_waveforms.swapaxes(1, 2)

# retrieve spike frames (sample_indices for detected waveforms) from reader
spike_frames = reader.get_spike_timestamps()        ##### CONFIRM THIS!!! #####
sampling_frequency = 10000                          
nbefore = 15                                        
recording_snippets = si.NumpySnippets(float_waveforms, spike_frames, sampling_frequency, nbefore=nbefore)

# since this object is in-meory, we have to save it to a folder in order to use docker
recording_snipptes = recording_snippets.save(folder="rec_snippets")

# run sorter
sorting_waveclus_snippets = si.run_sorter('waveclus_snippets', recording_snipptes,
                             output_folder='results_waveclus_snippets', docker_image=True,
                             verbose=True)
print(sorting_waveclus_snippets)
```
which prints:
```
WaveClusSortingExtractor: 1 units - 1 segments - 10.0kHz
  file_path: /home/alessio/Documents/Codes/spike_sorting/spikeinterface/sandbox/results_waveclus_snippets/times_results.mat
```

To use the docker image, you have to checkout this PR: #1171 

I think this fixes your issue :) closing!

mfvd, 2022-12-29:
Thank you @alejoe91 No error now! 

However, the code seems to be stuck (~45min) with the following output ```Docker: pulling image spikeinterface/waveclus-compiled-base:latest``` I'm quite new to Docker and I'm not sure it's working as it should.

To get the Docker image I used 
```Python
# run sorter
sorting_waveclus_snippets = ss.run_sorter('waveclus_snippets', recording_snipptes,
                             output_folder='results_waveclus_snippets', docker_image="spikeinterface/waveclus-compiled-base:latest",
                             verbose=True)
```

There is also an error with the number of channels. The recording has 120ch. I confirmed the file structure with [this code](https://www.neuroexplorer.com/docs/guides/readwritefiles.html) and indeed there is a mismatch.

Any idea of what might be going wrong?

ferchaure, 2022-12-29:
Hi, I didn't have time to check your data but waveclus is not build for recordings with high number of channels (and maybe is crashing for that reason). Can you run the code separating the channels in neighborhoods? It was developed for single channels and tetrodes

alejoe91, 2022-12-29:
@mfvd for the wrong channels, can you open an issue in NEO?

The first step of the docker run is to pull the image from [dockerhub](https://hub.docker.com/u/spikeinterface). This can take a while as MATLAB-based images can be over 5GB!

alejoe91, 2023-01-26:
@mfvd any update on this?

alejoe91, 2023-03-27:
Closing for inactivity, feel free to repoen

closed 2023-03-27 by alejoe91