SpikeInterface/spikeinterface #1178: Error in the curation step

kshtjkumar, 2023-01-01:
I am working on the .rhs neural data file. Following your pipeline I was able to do most of the steps but, on the Curation step when I run :
we_curated = we.select_units(keep_units_ids, new_folder = "waveform")

I get an error: 'TemplateMetricsCalculator' object has no attribute 'template_metrics".

[image]

Can someone please help with the installation of  this while from the source ? I am relatively new to this :) 

Thank you.

alejoe91, 2023-01-03:
You can find the instructions here: https://spikeinterface.readthedocs.io/en/latest/installation.html :)

samuelgarcia, 2023-01-03:
Yes this should be fix in the master.
See also this for beginners :  https://github.com/SpikeInterface/spikeinterface/tree/master/installation_tips

closed 2023-01-04 by alejoe91