SpikeInterface/spikeinterface #1179: error in opening sorted results with Phy

yinrocky, 2023-01-01:
The sorted results were exported to phy, when loading the results to phy. The following error message shown up. 
I'm running Ubuntu 22.04 LTS.  I tried to downgrade the pyqtwebengine and pyqt5 version to 5.12, but doesn't work. In windows 10, all works fine.
Many thanks!
[image]

alejoe91, 2023-01-02:
Hi, Phy installation is quite tricky with the QT dependencies. Can you open an issue on the Phy GitHub page since that is an external package?

yinrocky, 2023-01-02:
> Hi, Phy installation is quite tricky with the QT dependencies. Can you open an issue on the Phy GitHub page since that is an external package?

Okay, I see. thanks

closed 2023-01-04 by alejoe91