SpikeInterface/spikeinterface #1181: preprocessing filtering overflow memory

rat-h, 2023-01-03:
A recording: 128 channels, int16 at 30000Hz sampling, 1014s - total duration.  The binary file is about 7.3 Gb.

Code which should high pass filter this recording is below:
```python
from matplotlib.pyplot import *
import scipy
import spikeinterface as si  # import core only
import spikeinterface.extractors as se
import spikeinterface.sorters as ss
import spikeinterface.comparison as sc
import spikeinterface.widgets as sw
from spikeinterface.preprocessing import filter as spfilter 
# (common_reference, remove_artifacts, preprocesser_dict)
from probeinterface import read_probeinterface

recording = se.BinaryRecordingExtractor('continuous.dat',30000,128,'int16')
prob = read_probeinterface('A4x32-Poly2-5mm-23s-200-177-after-mapping.json').probes[0]
recording.set_probes(prob)
print(recording)

recording_hp = spfilter(recording,btype='highpass',band=300)
print(recording_hp)

fs = recording.get_sampling_frequency()
rawtr = recording.get_traces(segment_index=0)
print("Extracted RAW")
f_raw, p_raw = scipy.signal.welch(rawtr[:,0], fs=fs)
print("Done RAW")
hptr  = recording_hp.get_traces(segment_index=0)
print("Extracted Highpassed")
f_hp , p_hp  = scipy.signal.welch(hptr[:,0], fs=fs)
print("Done Highpassed")
semilogy(f_raw, p_raw, f_hp, p_hp)
show()
```
Output
```
python read-data.py 
BinaryRecordingExtractor: 128 channels - 1 segments - 30.0kHz - 1014.060s
  file_paths: ['XXXXXXXXXXX/spikeiterface/continuous.dat']
FilterRecording: 128 channels - 1 segments - 30.0kHz - 1014.060s
Extracted RAW
Done RAW
Killed
```

Filtering, for some reason, fills up all my 64Gb memory. Even if we assume that filtering converts everything into arrays with `float` precision, it should be 30-40 Gb of memory.

**Questions**
1. How to avoid memory overflow? 
2. Is it possible to run filtering in parallel?
3. Is there any way to do filtering on GPU (like in Kilosort)?

**A More general questions**
- How to make a "synthetic"  recording? Say, I can filter channel by channel in parallel with multiprocessing. How to combine filtered channels back together?
- I notice a `get_preferred_mp_context()` function of the `BaseExtractor` class and a set of tools in the `job_tools.py`  file, but it isn't clear to me how to use them. Any examples?

samuelgarcia, 2023-01-04:
Hi.
To set the probe you should do `recording = recording.set_probe(probe)`.

About the memory overflow : a key concept in spikeinterface is to not load in memory the entire traces because very often it is impossible.
This is why `recording_hp = spfilter(recording,btype='highpass',band=300)` is lazy.
You should never get all trace at once on this object.

For instance if you want to save back to another binary file you can do
```python
job_kwargs = {'n_jobs' : 5, 'chunk_duration':'1s', 'progress_bar':True}
write_binary_recording(recording_hp, file_paths='/path/to/new_file', **job_kwargs)
```
In that case, 5 worker will read in parrallel 1s chunk, then compute the preprocessing and then write to binary.
This is not emory consuming.

Extracting waveform, detecting peak, ... have the same "job" mechanism with this `job_kwargs`

Also importantly when you want to look at traces you can do :
```python
plot_timeseries(recording_hp, backend='ipywidget')
```
in that case the displayed traces are extrated and computed on the fly.

So you never need to load all the traces in memory!!

rat-h, 2023-01-04:
Hi @samuelgarcia,
Thank you for the explanations! They help a lot to get my head around.
I got that I can do `in_place=True` for `set_prob` to modify the `recording`.

However, I still have a question. So, my original file (binary file) is on the disk. It seems I have to save `recording`, which is obtained by `recording = se.BinaryRecordingExtractor`  to the disk through `write_binary_recording`. This step is needed to set `n_jobs` and  `chunk_duration` and enable parallelization and chunk splitting. But it actually doubles the size of data on the disk. Is it possible to create a `recording` object which will use an existing file and have all information about jobs and chunks? Something like that
```python
recording = si.Recording()
recording.binaryfile = path to the file
recording.nchannels=128
recording.samplerate=30000
recording.format = 'int16'
recording.set_probe(.....)
recording.n_jobs=40
recording.chunk_size='600s'
```

samuelgarcia, 2023-01-10:
Hi.
1. Settings a posteriori attributes to recording is bad in spikeinterface. All attributes must created in init this needed for parralell computing.
2. Attaching job params to a recording is a non-sens in spikeinterface. This job params are needed for the job executor when doing a task (writing, detecting peak, extracting wavefroms) and not to describe a state of a recording (num channel, sampling rate, dtype...).

alejoe91, 2023-01-10:
@rat-h just to add on @samuelgarcia comment. In general, you should never extract all traces at once (`recording.get_traces()`). This will for sure unnecessarily fill up your RAM, in particular when you are doing preprocessing, because the actual memory usage will increase dramatically due to the processing buffers.

In your case, I believe you want to estimate the spectrum of the signal. To do so, given the high sampling frequncy, you don't need the full traces! One minute would be more than enough. So here is how you would do it:

```
start_frame = 0
end_frame = int(60 * recording.sampling_frequency)
rawtr = recording.get_traces(segment_index=0, start_frame=start_frame, end_frame=end_frame)
print("Extracted RAW")
f_raw, p_raw = scipy.signal.welch(rawtr[:,0], fs=fs)
print("Done RAW")
hptr  = recording_hp.get_traces(segment_index=0, start_frame=start_frame, end_frame=end_frame)
print("Extracted Highpassed")
f_hp , p_hp  = scipy.signal.welch(hptr[:,0], fs=fs)
print("Done Highpassed")
semilogy(f_raw, p_raw, f_hp, p_hp)
show()
```

samuelgarcia, 2023-01-11:
Maybe I missunderstaood your question.

Maybe what you want to to "dump" your recording. In short saving the description without saving the "binary".
This operation is called dump in spikeinterface.

```python
recording = si.BinaryRecordingExtractor('continuous.dat',30000,128,'int16')
prob = read_probeinterface('A4x32-Poly2-5mm-23s-200-177-after-mapping.json').probes[0]
recording.set_probe(prob,in_place=True)

recording_dump_to_json('/path/to/myrec.json')

# and then later

rec = si.load_extractor('/path/to/myrec.json')

```

Thius json file will conrtain a description of your recording without any copy.
This also work for pre process recording!

rat-h, 2023-01-12:
Thank you for the reply. Here is what I want exactly - `tridesclous`'s interface
```python
dataio = DataIO(dirname='tredic-workdir')

# feed DataIO
dataio.set_data_source(
    type='RawData',
    filenames=['continuous.dat'],
    dtype='int16',
    sample_rate=30000.0,
    total_channel=128,
    bit_to_microVolt=None)
dataio.set_probe_file('A4x32-Poly2-5mm-23s-200-177-after-mapping.prb')
```
Note this code doesn't copy data in the `tredic-workdir` folder, just creates bench of json files with properties/parameters.

I actually want something like that in the spikeinterface. So it **will not** copy my original data into a new directory as my code above does. With 2.5 Tb a day of recordings, space is critical, and each reading/writing producer slows everything significantly.

alejoe91, 2023-01-12:
If the recording is binary, most sorters won't need to copy it. If it's another format it's required because the sorter need the data in a certain format..

rat-h, 2023-01-12:
@alejoe91 Correct! How to create a recording object (like the `dataio` object in the `tridesclous` example) in the spikeinterface?
The only way I've found is to read binary data and then save it.
```
recording = si.BinaryRecordingExtractor('continuous.dat',30000,128,'int16')
prob = read_probeinterface('A4x32-Poly2-5mm-23s-200-177-after-mapping.json').probes[0]
recording.set_probe(prob,in_place=True)

recording_raw = recording.save(folder="raw", n_jobs=16, total_memory="2G", progress_bar=True,chunk_duration='1m')
```
But this is horribly inefficient.

alejoe91, 2023-01-12:
If you don't want to preprocess, then you can directly pass the object (which is already binary) to the sorter. Some sorters (like Kilosort) will not re-save it because they need a Binary recording as input. If you want to "save" it an load it back, including the probe info, you can dump it to JSON as Samuel suggested

rat-h, 2023-01-12:
Please give me an example. Say you have a binary `int16' file, with 30 kHz sampling rate and 128 channels. Create a `recording` object that will refer to the binary file and have all the required parameters of the recording, like in the example for `tridesclous` above?

rat-h, 2023-01-12:
I have a binary file and the recording parameters (`int16`,128 channels), please give me an example how to create an `recoding` object from scratch, so that I can preprocess it and do everything else.

alejoe91, 2023-01-13:
What you have is perfectly correct:
```
recording = si.BinaryRecordingExtractor('continuous.dat',30000,128,'int16')
prob = read_probeinterface('A4x32-Poly2-5mm-23s-200-177-after-mapping.json').probes[0]
recording.set_probe(prob,in_place=True)
```
Then you can pre-process the recording and continue with the analysis. 

(If you want to save it to disk, I recommend using `chunk_duration="1s"` instead!)

rat-h, 2023-01-14:
> What you have is perfectly correct:

I don't think so. These 3 lines load the binary file into memory and then any preprocessing overflows memory (see original issue). Only if I save this object and start to process 'saved' object (`recording_raw` my workaround), memory overflow can be fixed. 

In tridesclous example: (1) the binary file is not loaded into memory as in 3 lines example above, (2) it does not copy the whole binary file into the working directory, like my workaround does, and (3) it works with chunks as saved object does in the spikeinterface.

alejoe91, 2023-01-14:
> > What you have is perfectly correct:
> 
> I don't think so. These 3 lines load the binary file into memory and then any preprocessing overflows memory (see original issue). Only if I save this object and start to process 'saved' object (`recording_raw` my workaround), memory overflow can be fixed.
> 

I think there was a misunderstanding. Let me try to clarify:

The memory overflow happens when you call `recording_hp.get_traces(segment_index=0)`, which attempts to load all traces in memory and filter them all in a block. This is something you should avoid and loading all traces in memory will fail for long recordings even with parallel processing. 

Can I ask you again  you are trying to achieve?
If you are trying to estimate the spectrum, then you should only load a small portion of the data, say one minute.
If you are trying to spike sort, then you can directly pass the object to the sorters.

> In tridesclous example: (1) the binary file is not loaded into memory as in 3 lines example above, (2) it does not copy the whole binary file into the working directory, like my workaround does, and (3) it works with chunks as saved object does in the spikeinterface.

The 3 lines that I mentioned (which are also what you suggested) achieve (1) and (2), meaning that the `recording` object is just a data retriever, which reads the traces from disk *lazily* only when you call `get_traces()`. The `tridesclous.dataio` object will also fail if you attempt to load all traces at once, no matter the chunking.

I hope I made myself clearer :)

rat-h, 2023-01-14:
@alejoe91 Thank you so much for the answer.

> Can I ask you again you are trying to achieve?

I have an experimental lab with ~2.5 TB of data in a day of recording. That are very unusual neurons, so standard methods like kilosort 2.5/3.0 do not sort spikes well.
I'm trying to find a good pipeline of methods to make spike sorting acceptable. From this point of view, the spikeinterface is perfect! So, I'm constantly changing preprocessing, sorting, and postprocessing to improve sorting results on a big database of recordings.

Now, do I get here: I tried an example from the spikeinterface tutorial, and it overflowed my memory. I misinterpreted the response from @samuelgarcia as the initial recording object must be saved to enable chunk processing, and that doubled required space on the disk and made me unhappy. Now it is clear where was the problem, and why it happened.

Again, thank you so much for the last clear explanations, which pointed out where I slipped off.

alejoe91, 2023-01-14:
No worries! :) Thank you for the support!

rat-h, 2023-01-16:
Oh, it seems I have to reopen the issue
Here is the code
```python
import spikeinterface.full as si
from probeinterface import read_probeinterface
recording = si.BinaryRecordingExtractor('continuous.dat',30000,128,'int16')
prob = read_probeinterface('probes/A4x32-Poly2-5mm-23s-200-177-after-mapping.json').probes[0]
recording.set_probe(prob,in_place=True)

tds2_sorting = si.run_sorter(
        sorter_name='tridesclous2',
        recording=recording, 
        output_folder="tds2-workdirectory")
tds2_sorting.save(folder="tds2-sorting")
tds2_we = si.extract_waveforms(
    recording, tds2_sorting, 'tds2-waveforms',
    max_spikes_per_unit=500,
    ms_before=5, ms_after=8,
    n_jobs=16, total_memory='10G',
    progress_bar=True,
)
```
and during execution it indicates a warning because probe was not found
```
.../lib/python3.10/site-packages/spikeinterface/core/baserecordingsnippets.py:212: UserWarning: There is no Probe attached to this recording. Creating a dummy one with contact positions
  warn(
Setting 'return_scaled' to False
....
```

However, if I save this recording (as in my workaround above) this message does not appear.

```python
import spikeinterface.full as si
from probeinterface import read_probeinterface
recording = si.BinaryRecordingExtractor('continuous.dat',30000,128,'int16')
prob = read_probeinterface('probes/A4x32-Poly2-5mm-23s-200-177-after-mapping.json').probes[0]
recording.set_probe(prob,in_place=True)

recording  = recording.save(folder="tds2-raw", n_jobs=16, total_memory="2G", progress_bar=True,chunk_duration='1m')

tds2_sorting = si.run_sorter(
        sorter_name='tridesclous2',
        recording=recording, 
        output_folder="tds2-workdirectory")
tds2_sorting.save(folder="tds2-sorting")
tds2_we = si.extract_waveforms(
    recording, tds2_sorting, 'tds2-waveforms',
    max_spikes_per_unit=500,
    ms_before=5, ms_after=8,
    n_jobs=16, total_memory='10G',
    progress_bar=True,
)

```

```
write_binary_recording with n_jobs = 16 and chunk_size = 488281
write_binary_recording: 100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 63/63 [00:03<00:00, 17.54it/s]
Setting 'return_scaled' to False
```

Any thoughts?

samuelgarcia, 2023-01-19:
Hi Ruben,
this " UserWarning: There is no Probe attached to this recording. Creating a dummy one with contact positions" was fixed I think.
Are you using the master or a release ?

rat-h, 2023-01-20:
I'm using from pypi installed by pip.

samuelgarcia, 2023-01-20:
You should try install from source with github master.
The last reelase was in september. We have fixed a lot of bugs since.

We plan a new relase mid frebruary.

closed 2023-01-20 by rat-h