SpikeInterface/spikeinterface #1184: Struggles isolating tetrode data recorded with shuttle drives and the OpenEphys system

ThePuppetShow, 2023-01-06:
Hey! I've done some recordings using 64ch Shuttle Drives in mice, with the Open Ephys recording system. My data is stored as .continuous files, and consists of 16 tetrodes that need to be spikesorted as "isolated" groups of 4. 

I'm struggling to determine how to properly design the "geometry" of my recordings to isolate clusters within tetrodes. I've managed to install and run the code in python, and sorted the data with mountainsort4 (through docker). 

In addition to grouping my data in tetrodes, I also need to parse a channel map from OpenEphys, as indices 1-64 are scrambled by connecting the drive to the computer. The channel map was set in Open Ephys while recording, but to my knowledge the .continuous files are stored without this map being considered (I can add that we used the 0.4.6 release of the OE GUI). 

When I look at the recording in the spikeinterface-gui, I can see my 16 tetrodes on a line, with correct channel names. However, the clusters that appear are not always surrounding the tetrodes where they are expected to be. Sometimes they are, but some, like the  cluster close to tetrode [41, 43, 45, 46] was picked up by a tetrode far away [49, 51, 53, 55] (see attachment). I also noticed that some waveforms are inverted, although this was not seen in the OE GUI (see attachment). I'm also not sure if spikes that are visible on multiple channels in a tetrode are recorded by the combined waveforms within the tetrode, or by one channel at a time. 

I started using python very recently, so hopefully I can manage to convey my challenges (and understand what to do). I've added my code below, which is largely based on the spikeinterface guidelines. I also uploaded some images that'll hopefully make my confusion more clear.

Spikeinterface seems like an amazing tool, and I love how compatible it is with all manners of sorters. I hope I'll be able to utilise it : ) 

Best
Jo

```
## Import packages
import matplotlib.pyplot as plt
plt.rcParams['figure.figsize'] = (15, 5)
import numpy as np
from numpy import *

import spikeinterface.full as si

from probeinterface import Probe, ProbeGroup
from probeinterface.plotting import plot_probe, plot_probe_group
from probeinterface import generate_tetrode

## Ask for directory with recording session
import promptlib
prompter = promptlib.Files()
local_path = prompter.dir()
print(local_path)

## Read OE session
recording = si.read_openephys(local_path, stream_name='Signals CH')

## Create tetrode file

## Channel map Giles
IGiles = [58, 64, 62, 60, 63, 61, 59, 57,
        55, 53, 51, 49, 56, 54, 52, 50,
        47, 45, 43, 41, 39, 37, 35, 33,
        40, 38, 36, 34, 25, 27, 29, 31,
        4, 6, 8, 2, 17, 19, 21, 23,
        9, 11, 13, 15, 1, 3, 5, 7,
        10, 12, 14, 16, 18, 20, 22, 24,
        26, 28, 30, 32, 48, 46, 44, 42] 

## Subtract 1 to fit python indexing
IdxA = IGiles[:]
for i in range(len(IdxA)):
    IdxA[i] -= 1
print(IdxA)
device_channel_indices=IdxA

## Create probegroup
probegroup = ProbeGroup()
for i in range(16): # We have 16 tetrodes
    tetrode = generate_tetrode()
    tetrode.move([i * 300, 0]) # Make lots of space between tetrodes
    probegroup.add_probe(tetrode)

probegroup.set_global_device_channel_indices(array(device_channel_indices)) ##Tetrodes arranged based on channelmap
recording.set_probegroup(probegroup, group_mode='by_probe', in_place=True)

## Preprocess
recording_cmr = recording
recording_f = si.bandpass_filter(recording, freq_min=300, freq_max=6000)
recording_cmr = si.common_reference(recording_f, reference='local', operator='average', ref_channel_ids=['26', '28', '30', '32'], local_radius=(1, 80)) 

## Channels 26, 28, 30, 32 belong to the most silent tetrode this recording. 

## Create path for sorting
import os
sortingPath=(f"{local_path}{'/SortedData'}")
if not os.path.exists(sortingPath):
    os.makedirs(sortingPath)

sorting_MS = si.run_mountainsort4(recording=recording_cmr, detect_threshold=4, output_folder=sortingPath,
    docker_image=True)
print(sorting_MS)

## Extract waveforms
waveform_folder = (f"{local_path}{'/Waveforms'}")
we = si.extract_waveforms(
    recording_cmr, sorting_MS, waveform_folder,
    max_spikes_per_unit=500,
    ms_before=1.5, ms_after=2.5,
    n_jobs=10, total_memory='500M',
    progress_bar=True,
)

## Principal components
pc = si.compute_principal_components(we,
    n_components=5,
    mode='by_channel_local',
    `whiten=True)`
```

[image]
Here you see some clusters next to a tetrode. the spikes are clear in the traces, but inverted in some channels for some weird reason. 

[image]
Here is the same tetrode from the OE GUI (waveforms are inverted - but the same in all 4 channels)

[image]
This cluster is next to one tetrode, but looking at the average waveforms, it is taken from a tetrode far away.

samuelgarcia, 2023-01-06:
Hi Joachim,
First : when you have several tetrode you have several to do the sorting:
 1. Your case. Make a fake geometry where tetrode are far away. And sort all the recording at once.
 2. Split virtually your recording in sub recording and sort every tetrode independantly. [See this doc ](https://spikeinterface.readthedocs.io/en/latest/modules/sorters/plot_3_sorting_by_group.html#sphx-glr-modules-sorters-plot-3-sorting-by-group-py)

Many people prefer the case 2. I recommand the case 1 as you do because in case 2 you need to view the results tetrode by tetrode. For the case 1 montainsort4 is maybe not the best choice. Maybe if you really want maintainsort4 then I would go for case 1 (splitting by group). So maybe try other sorters.

Your code looks totally good for beginners in python.

Note that the filter in open-ephys and in spikeinterface are note the same : OE is doing forward filtering SI is doing foward-backward (filtfilt). This is why the waveform do not looks the same.

Your main problem I think is the channel mapping!
If you have clusters spread on several tetrode this is a bad sign..

The `IGiles` variable have to read as follow:
  * The first contact of the first tetrode corespond to channel 58 (zeros based in open ephys)
  * The second contact of the first tetrode corespond to channel 64 (zeros based in open ephys)
...
 * The fourth contact of the 16th tetrode correspond to channel 42  (zeros based in open ephys)

Unless I am doing a mistake I do not see any 0 in `IGiles`.
Here the mapping is zero-based (0-63)
So my first guess would be : Gilles is a matlab guy and he gave you the mapping in one-based (1-64)
So maybe -1 would fix the stuff.

Otherwise carrefully check carrefully the mapping. Note th openephys is displaying one-based!!

And also : thank you for using openephys-gui, you are few users!

ThePuppetShow, 2023-01-10:
Hey and thanks for the thorough and helpful reply! 

I tried to split the recording as you suggested and that worked out well. I need to process tetrodes individually, but there is no "interference" between tetrodes which is what I wanted. 

So thanks a lot!

```
recordings_prep=list()  ## This is stored as a list because it can contain 1 or more folders with recording data
for i in range(len(recording_path)):
    recordings[i].set_probegroup(probegroup, group_mode='by_probe', in_place=True)
    # Preprocess
    recording_cmr = recordings[i]
    recording_f = si.bandpass_filter(recordings[i], freq_min=300, freq_max=6000)
    recording_cmr = si.common_reference(recording_f, reference='local', operator='average', ref_channel_ids=ref_ch, local_radius=(1, 80))
    recordingsplit = recording_cmr.split_by(property='group')
    recordings_prep.append(recordingsplit)

## Loop through folders
for s in range(len(recording_path)): ## Here, s corresponds to separate folders with recording data
    print(f"{'Folder '}{s}")
    rec_temp = recordings_prep[s]

    ## Loop through tetrodes
    for i in range(16):  ## "i" represents different tetrodes in the given recording session
        print(f"{'Tetrode '}{i}")

        tetName = str(i)
        sortingPath = (f"{recording_path[s]}{'/SortedData'}{tetName}")
        if not os.path.exists(sortingPath):
            os.makedirs(sortingPath)

        recording_tt = rec_temp[i]
        sorting_SC = si.run_spykingcircus(recording=recording_tt, output_folder=sortingPath, docker_image=True)
        print(sorting_SC)

        #### Extract waveforms
        waveform_folder = (f"{recording_path[s]}{'/WaveformsSC'}{tetName}")
        we = si.extract_waveforms(
             recording_tt, sorting_SC, waveform_folder,
             max_spikes_per_unit=5000,
             ms_before=1.5, ms_after=2.5,
             n_jobs=10, total_memory='500M',
             progress_bar=True)

        # and optionally compute principal component
        pc = si.compute_principal_components(we,
            n_components=5,
            mode='by_channel_local',
            whiten=True)

        # we = si.WaveformExtractor.load_from_folder(waveform_folder1)
        from spikeinterface.exporters import export_to_phy
        phy_folder = (f"{recording_path[s]}{'/Phy '}{i}")
        export_to_phy(we, phy_folder)

```

The channel map was actually correct. There's a piece of code that subtracts one from the Giles index. I kept the original index indexed from 1 just to match my notes. 

I wanted to ask though, does the spikeinterface pipeline consequently use indexes from 0? When choosing channels for my common median reference, I get an error stating that "ValueError: '0' is not in list" if I choose 0 among my ref channels. Does this mean that the reference channels start indexing from 1?

On a side note: I did run into an error with this setup when trying to use Klusta on unsorted. This is not crucial for my pipeline, but I thought I'd mention it. I get the following error: 

```
spikeinterface.sorters.utils.misc.SpikeSortingError: Spike sorting in docker failed with the following error:
b'Traceback (most recent call last):\n  File "/TempAnalysisFolder/GilesS7/2022-11-09_16-59-57_JSG_GilesS7_BL1/in_container_sorter_script.py", line 4, in <module>\n    from spikeinterface.sorters import run_sorter_local\n  File "/usr/local/lib/python3.7/site-packages/spikeinterface/sorters/__init__.py", line 2, in <module>\n    from .sorterlist import *\n  File "/usr/local/lib/python3.7/site-packages/spikeinterface/sorters/sorterlist.py", line 1, in <module>\n    from .combinato import CombinatoSorter\n  File "/usr/local/lib/python3.7/site-packages/spikeinterface/sorters/combinato/__init__.py", line 1, in <module>\n    from .combinato import CombinatoSorter\n  File "/usr/local/lib/python3.7/site-packages/spikeinterface/sorters/combinato/combinato.py", line 10, in <module>\n    from spikeinterface.extractors import CombinatoSortingExtractor\n  File "/usr/local/lib/python3.7/site-packages/spikeinterface/extractors/__init__.py", line 4, in <module>\n    from .bids import read_bids_folder\n  File "/usr/local/lib/python3.7/site-packages/spikeinterface/extractors/bids.py", line 4, in <module>\n    import pandas as pd\nModuleNotFoundError: No module named \'pandas\'\n'
```

Weird thing is that I do have pandas version 1.4.4. I also have python 3.9.15 installed, but this error opens an init file within a python 3.7 folder, I'm guessing this is happening within the docker system somehow? I know Klusta needs an older version of python. Klusta does work on data I already sorted with Spyking Circus, but I assumed it would work here as well since it's listed as compatible with raw data (https://spikeinterface.readthedocs.io/en/latest/supported_formats_and_sorters.html#sorted-data-formats).

alejoe91, 2023-04-27:
@ThePuppetShow can we close this?

ThePuppetShow, 2023-04-27:
Yes! Thanks a lot for your help. I truly appreciate it.

Best
J

closed 2023-06-12 by alejoe91