SpikeInterface/spikeinterface #1187: question: interpolation for WaveformExtractor.run_extract_waveforms

ferchaure, 2023-01-07:
Hi Guys, after a few months I'm going back to spike sorting :smile:  . I'm comparing sessions using WaveformExtractor, but the waveforms are not perfectly align (again this is a mainly single channel issue) and that increase the noise between classes.

What do you think about an optional parameter in run_extract_waveforms to interpolate the waveforms and realign with the peak?
Something similar to preprocessing/align_snippets

samuelgarcia, 2023-01-10:
Hi Fernando.
Welcome back!

My intuitive feeling would be to not complexify to much the waveform extractor itself and so not adding interpolation in it.
A possibility easy to implement would be to make a `BaseWaveformExtractorExtension` that would do that.
The cost is big because it double the size of waveforms but for one channel I guess this is OK.

A main difficulty will be to extend this concept to multi channel. I guess we want to interpolate all channel but to align the main channel only. No ?

alejoe91, 2023-01-10:
Hi @ferchaure 

Alternatively, we could have in the `postprocessing` module a `AlignedWaveformExtractor(WaveformExtractor)` class that overriedes the `get_waveforms` and `get_template` by applying the correct shift by unit (similar to the `align_sorting`).

What do you guys think?

ferchaure, 2023-01-10:
Then a class like @alejoe91 proposed will work for everyone.  Do you think it should be necessary to correct the frame for each spike inside the sorting? (`align_sorting` corrects by unit, not by spike; doesn't it?)

I'll do a pull request, and we can keep talking about all the required methods.

@yger, your idea is: calculate templates, use the templates to improve spike alignment and then recompute templates again? then the original WaveformExtractor should have at least 2 extra samples on the borders (?).

If calculating a WaveformExtractor on ram is possible, then using the class constructor to save only the align/interpolated WaveformExtractor will be perfect.

yger, 2023-01-10:
Yes, that is exactly what I have in mind. This means that the first WaveformExtractor has indeed larger bounds compared to the one that would result and be aligned. You should provide bounds such as (ms_before, ms_after) plus epsilon (in ms), to get at the end an aligned WaveformExtractor os size (ms_before,ms_after). However this should use multiple jobs, because on a spike by spike basis, such a procedure will be slow

ferchaure, 2023-04-06:
I'm giving up on this pull request, for my needs I ended up using a fake waveformextractor from Waveclus outputs and I don't have more time to develop that project. I still think that this is important but I'm not sure about the proper way to implement it and record that the interpolation+alignment was made inside the waveformextractor, it will affect the quality metrics and the reproducibility of the results.

zm711, 2024-03-25:
I'll close this as we have moved to SortingAnalyzer. But feel free to open a new if something like this would be beneficial for the SortingAnalyzer in the future.

closed 2024-03-25 by zm711