SpikeInterface/spikeinterface #1188: issues with saving a recording to disk

LaurenzMuessig, 2023-01-09:
Hi,

I have issues with saving the recording from the extractor to disk either using _recording.save()_ or _write_binary_recording()_. 

My data is a neuropixel 1.0 recording - not too large, two 20 minute trial that I concatenated (~55GB) beforehand. Then I do a bunch of preprocessing (bp filtering, phase_shift, common ref) and some motion correction. When I want to write the processed recording to disk (to run a containerised sorter), I can't seem to find a set of parameters that writes the file in a reasonable amount of time - either it's super slow or it crashes with a MemoryError

print(recording)
SpikeGLXRecordingExtractor: 384 channels - 1 segments - 30.0kHz - 2449.685s

e.g.
_write_binary_recording(recording,file_paths=os.path.join(params['dirOut'],'tempData'),n_jobs=-1,chunk_size=30000,verbose=True,progress_bar=True)_ 
results in:
MemoryError: Unable to allocate 644. GiB for an array with shape (30000, 30000, 384) and data type int16

Not sure if there is something that I don't understand but writing a similar lengths bin file to disk using standard linux commands takes maybe ~5-10 minutes on the machine I am using. Maybe someone could give me a nudge in the right direction? Any help is highly appreciated.  

Cheers

Laurenz

samuelgarcia, 2023-01-10:
Hi. This shape error looks like a bug we fix already.
Could you update with the master sources ?

LaurenzMuessig, 2023-01-10:
Hi @samuelgarcia ,
Thanks for your quick response. I am using the latest version I think. The problem only occurs when trying to save the motion corrected recording (I use _monopolar_triangulation_ and _decentralised_registration_). 

Cheers

samuelgarcia, 2023-01-11:
Your pip have strange behavior.

I would recommand:
 * to be sure to work in an isolated environement for instance using venv:
   * `python3.10 -m venv /path/to/my/si_env`
   * `/path/to/my/si_env/bin/activate`
 * to update pip by doing `pip install --upgrade pip`

LaurenzMuessig, 2023-01-11:
I am working in a virtual environment. Not sure why pip would pull on old version. In any case, I think I have solved my issue. 
When trying to do:
_python setup.py install_

this seems to depend on the exact version of setuptools. I had the most recent one in my venv and that produced am error (a different one to the one I posted above). I then downgraded setuptools to a version below 62.0 (didn't check the dependencies beforehand) and this yields the error above. After upgrading to 62.1 it seems to be working as it should. Thanks for your help @samuelgarcia .

closed 2023-01-11 by LaurenzMuessig