SpikeInterface/spikeinterface #1194: Error when loading OpenEphys .continuous files, but only in some recording sessions

ThePuppetShow, 2023-01-10:
Hey! 

I’m In the process of analysing data recorded with the Open Ephys system using 16 tetrode shuttle drives (64ch), but I’m having problems loading .continuous files in some recordings (while other sessions have no issues loadint at all). When I execute “recordings.append(si.read_openephys(recording_path[i], stream_name='Signals CH'))” I get the following error:

```
Traceback (most recent call last):
  File "C:\Users\joach\anaconda3\envs\si_env\lib\site-packages\IPython\core\interactiveshell.py", line 3433, in run_code
    exec(code_obj, self.user_global_ns, self.user_ns)
  File "<ipython-input-12-314ef08b03de>", line 53, in <module>
    recordings.append(si.read_openephys(recording_path[i], stream_name='Signals CH'))
  File "C:\Users\joach\anaconda3\envs\si_env\lib\site-packages\spikeinterface\extractors\neoextractors\openephys.py", line 220, in read_openephys
    recording = OpenEphysLegacyRecordingExtractor(folder_path, **kwargs)
  File "C:\Users\joach\anaconda3\envs\si_env\lib\site-packages\spikeinterface\extractors\neoextractors\openephys.py", line 56, in __init__
    NeoBaseRecordingExtractor.__init__(self, stream_id=stream_id,
  File "C:\Users\joach\anaconda3\envs\si_env\lib\site-packages\spikeinterface\extractors\neoextractors\neobaseextractor.py", line 43, in __init__
    _NeoBaseExtractor.__init__(self, block_index, **neo_kwargs)
  File "C:\Users\joach\anaconda3\envs\si_env\lib\site-packages\spikeinterface\extractors\neoextractors\neobaseextractor.py", line 25, in __init__
    self.neo_reader = get_reader(self.NeoRawIOClass, **neo_kwargs)
  File "C:\Users\joach\anaconda3\envs\si_env\lib\site-packages\spikeinterface\extractors\neoextractors\neobaseextractor.py", line 15, in get_reader
    neo_reader.parse_header()
  File "C:\Users\joach\anaconda3\envs\si_env\lib\site-packages\neo\rawio\baserawio.py", line 179, in parse_header
    self._parse_header()
  File "C:\Users\joach\anaconda3\envs\si_env\lib\site-packages\neo\rawio\openephysrawio.py", line 117, in _parse_header
    assert np.all(diff == RECORD_SIZE), \
AssertionError: Not continuous timestamps for 100_AUX1.continuous. Maybe because recording was paused/stopped.

```

I don’t remember pausing or stopping this recording, so I don’t think this is the reason for the error. I also don’t think the data is corrupted, as I am able to read it in my Matlab pipieline using OpenEphys’ own loading function “[data, timestamps, info] = load_open_ephys_data(inputdata)”. 

I get this error with a handful of recording sessions in my dataset, and I really need to get it to work but I have no clue how to fix this error. Any insight as to how it could be fixed would be very much appreciated. 

Best regards
Jo

samuelgarcia, 2023-01-11:
I guess you are using the old binary openephys format (that split all channels in one files).
This format is not the recommanded one.
You should use the new binary with one unique file.

I guess there a bug or you have a gap maybe because of a missing package.
You can solve your problem by moving out from the folder the "100_AUX1.continuous" which is not necessary for sorting.

You also submit this to python-neo.

alejoe91, 2023-01-26:
Hi Jo, 

Let's continue the discussion on the NEO side. Closing this one!

Cheers
Alessio

closed 2023-01-26 by alejoe91