SpikeInterface/spikeinterface #1195: While merging two .dat files, the system goes out of memory.

lmontelisciani, 2023-01-10:
I use the following script to merge two recordings done with the Neuropixel probes:

`import spikeinterface as si

first_probe_spike_start = si.BinaryRecordingExtractor(r'path_to_file\continuous.dat', sampling_frequency =30000, num_chan=385, dtype ='int16')
first_probe_spike_end   = si.BinaryRecordingExtractor(r'path_to_file\continuous.dat', sampling_frequency=30000, num_chan=385, dtype='int16')

first_probe_spike_merged = si.concatenate_recordings([first_probe_spike_start, first_probe_spike_end])

first_probe_spike_merged.save(format='binary', folder=r'C:\DATA_OpenEphys\Newresult')`

However, once I save it, the system says it can not allocate the file since it's too big (attached
[image]
). Does someone know why this happened?

Thanks in advance for any help.

samuelgarcia, 2023-01-11:
You need to chunk you saving
`first_probe_spike_merged.save(format='binary', folder=r'C:\DATA_OpenEphys\Newresult', n_jobs=8, chunk_duration='1s')`

lmontelisciani, 2023-01-11:
Thanks a lot, it worked!

closed 2023-01-11 by lmontelisciani