SpikeInterface/spikeinterface #1202: `cluster_si_unit_ids.tsv` from `export_to_phy` output is not updated properly in phy

a-zmz, 2023-01-13:
Hi,

In the output of `export_to_phy`, I get a `cluster_si_unit_ids.tsv` file, and it does not get  proper update with merge/split. As a result, I have some units that do not have `si_unit_id`, and units ids loaded from `sorting.unit_ids` do not correspond to `cluster_info.tsv`, although the total number of units are the same...

[image]

This causes a problem in subsequent analyses when I need to access units, I cannot be sure whether I am accessing the information of the unit that I intended to.

What is the function of this `cluster_si_unit_ids.tsv` file, is it necessary as a output of `export_to_phy? Can it be removed as it causes much confusion... Thanks!

alejoe91, 2023-01-14:
Hi @a-zmz 

The `si_unit_id` represents the unit id in SpikeInterface. The `id` column, instead, needs to be a linearly increasing `int`. If you reimport the phy data to SpikeInterface, the original id (for unmerged/unsplitted units) would be lost. Note that when loading back to SI (`read_phy`), new `si_unit_id`s are assigned automatically to units that don't have it!

So I recommend not to remove it, but if you are not planning to reimport to SpikeInterface, then it doesn't matter

a-zmz, 2023-01-16:
Hi Alessio thanks for quick replying.

It is still a bit confusing for me:
1. the results from running ks3 through si does not have `si_uni_id`, unit ids of sorting object from importing these results into si correspond to `cluster_id` in `cluster_info.csv`, i.e., uni ids in si are the same as cluster id. If this works for si without `si_uni_id` then `export_to_phy` output should also work?
2. i did try `read_phy` and `read_kilosort`, the sorting obj i got have the correct number of units. you mentioned that new `si_unit_id` will be assigned, but will the new assignment be saved/cached somewhere so i can map it to `cluster_id`? cuz the `cluster_info.tsv` is not updated...

Thanks!

alejoe91, 2023-01-18:
> Hi Alessio thanks for quick replying.
>
> It is still a bit confusing for me:
>
> 1. the results from running ks3 through si does not have `si_uni_id`, unit ids of sorting object from importing these [...]

That is only true for Kilosort, because it outputs linearly increasing unit ids as integers (which is what phy needs). Note that if you remove empty units, the phy units and spikeinterface units will not match anymore. Say you manually flag some units as noise and then you reload the phy folder in spikeinterface, the `si_unit_id` will make sure that the newly loaded `unit_ids` correspond to the original unit_ids.

> 2. i did try `read_phy` and `read_kilosort`, the sorting obj i got have the correct number of units. you mentioned that new `si_unit_id` will be assigned, but will the new assignment be saved/cached somewhere so i can map it to `cluster_id`? cuz the `cluster_info.tsv` is not updated...
> 
The new `si_unit_id` will appear as `unit_ids`. All available columns from phy will be loaded accordingly. 

I hipe this is clearer!

a-zmz, 2023-01-18:
hi Alessio, after `read_phy`, I i did get all units from the sorting object, but columns in phy are not updated, i.e., the file `cluster_si_unit_id.tsv` and `cluster_si_unit_ids.tsv` are not updated...

alejoe91, 2023-01-18:
yes the files are not updates, because it's a `read_*` function. Why do you need to update the `.tsv` files?

a-zmz, 2023-01-18:
lol right...
Because currently many of my subsequent analyses rely on `cluster_info.tsv`, and i index units using `cluster_id` in this file. I guess essentially i wonder if there is a way to map `si_unit_ids` to `cluster_id`?

alejoe91, 2023-01-19:
@a-zmz maybe the best option then would be to load all columns of the cluster_info as properties, so we are 100% sure that we are assigning them correctly. I'll make a PR so you can test it out

a-zmz, 2023-01-19:
@alejoe91 that sounds great thanks!

closed 2023-01-20 by samuelgarcia