SpikeInterface/spikeinterface #1206: 'TemplateMetricsCalculator' object has no attribute 'template_metrics'

vncntprvst, 2023-01-18:
Just wanted to point out this error keeps coming up. See issues [#1109](https://github.com/SpikeInterface/spikeinterface/issues/1109#issue-1465058069) and [#1178](https://github.com/SpikeInterface/spikeinterface/issues/1178#issue-1515724217). 
I believe it's been fixed in by [the commit](https://github.com/SpikeInterface/spikeinterface/pull/1112) on Nov 28, 2022, but the last pip release, 0.96.1, dates from Oct 27, 2022. So this error will happen for those who rely on a pip install, e.g. from a `requirement.txt` such as those provided in the tutorials.

vncntprvst, 2023-01-18:
The error stack, just in case:
```
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
Cell In[98], line 1
----> 1 we_curated = we.select_units(keep_unit_ids, new_folder="waveforms_cur")
      2 # we_curated = we.select_units(keep_unit_ids, new_folder=None, use_relative_path="waveforms_curated")

File ~\miniconda3\envs\si_env\lib\site-packages\spikeinterface\core\waveform_extractor.py:529, in WaveformExtractor.select_units(self, unit_ids, new_folder, use_relative_path)
    527 for ext_name in self.get_available_extension_names():
    528     ext = self.load_extension(ext_name)
--> 529     ext.select_units(unit_ids, new_waveform_extractor=we)
    531 return we

File ~\miniconda3\envs\si_env\lib\site-packages\spikeinterface\core\waveform_extractor.py:1124, in BaseWaveformExtractorExtension.select_units(self, unit_ids, new_waveform_extractor)
   1122 new_extension = self.__class__(new_waveform_extractor)
   1123 new_extension.set_params(**self._params)
-> 1124 new_extension_data = self._select_extension_data(unit_ids=unit_ids)
   1125 new_extension._extension_data = new_extension_data
   1126 new_extension._save()

File ~\miniconda3\envs\si_env\lib\site-packages\spikeinterface\postprocessing\template_metrics.py:53, in TemplateMetricsCalculator._select_extension_data(self, unit_ids)
     51 def _select_extension_data(self, unit_ids):
     52     # filter metrics dataframe
---> 53     new_metrics = self.template_metrics.loc[np.array(unit_ids)]
     54     return dict(metrics=new_metrics)

AttributeError: 'TemplateMetricsCalculator' object has no attribute 'template_metrics'
```

alejoe91, 2023-01-18:
Thanks @vncntprvst 

We are going to make a new release early February, including this fix! Thanks for bringing this up!

vncntprvst, 2023-01-18:
Sounds good!

closed 2023-01-18 by vncntprvst