SpikeInterface/spikeinterface #1212: Using older Npx probes

andrada08, 2023-01-20:
Hey! I am trying to use spikeinterface on some data acquired with openephys from Neuropixel 3A probes (version earlier than the 1.0s). It keeps telling me to use set_probe() because there are no probes in the recording - I think this is because it can't get what it needs from the settings file as these probes don't use the 'Neuropix-PXI plugin'. From my understanding, set_probe needs a bit of  setup but the geometry of these probes is identical to the 1.0s, is there a way for me to somehow use that?

alejoe91, 2023-01-26:
Hi @andrada08 

Probeinterface can only load probe designs from recent Open Ephys versions. What you can do is to set the probe externally with a `json` file from probeinterface that describes the NP1.0 geometry (see attachedzip file, that you need to unzip).

Then you can do the following:

```
import spikeinterface.extractors as se
import probeinterface as pi

recording = se.read_openephys(path_to_folder)
probe = pi.read_probeinterface("np1_standard.json")
recording = recording.set_probe(recording)
```
[np1_standard.zip](https://github.com/SpikeInterface/spikeinterface/files/10507665/np1_standard.zip)

andrada08, 2023-01-26:
Thanks so much, @alejoe91! This worked for me; instead of the last line, I did this:

`recording = recording.set_probes(probe)`

One more thing: I can't use `phase_shift` from preprocessing because `inter_sample_shift` it's not in the properties list. How can I set that?

alejoe91, 2023-01-26:
This way (assuming Phase3 has the same number of ADCs than NP1.0:
```
from spikeinterface.extractors.neuropixels_utils import get_neuropixels_sample_shifts

num_channels_per_adc = 12
sample_shifts = get_neuropixels_sample_shifts(self.get_num_channels(), num_channels_per_adc)
recording.set_property("inter_sample_shift", sample_shifts)
```

alejoe91, 2023-02-02:
@andrada08 I think this is solved. Please reopen or open a new issue should you have other questions/problems :)

closed 2023-02-02 by alejoe91

andrada08, 2023-02-02:
@alejoe91 yes this fixed the preprocessing issue I was having, thanks! 

Sorry I left this open, it's because sorting doesn't work and I have been trying to figure out why so I can ask a more specific question - but I don't really understand what the problem is. The error it gives is that `n_channels` is `None` but that doesn't make sense to me because `get_channel_locations()` gives me a list of things. I've attached the log file if that helps. 

[spikeinterface_log.zip](https://github.com/SpikeInterface/spikeinterface/files/10570109/spikeinterface_log.zip)