SpikeInterface/spikeinterface #1213: Make NwbRecordingExtractor accept electrode location coordinates labeled with 'x' besides 'rel_x'?

oaaij-gnahz, 2023-01-20:
Hi, 

I was using the `NwbRecordingExtractor` and the channel locations could not be read. 

It seems like on [Line 220-232](https://github.com/SpikeInterface/spikeinterface/blob/master/spikeinterface/extractors/nwbextractors.py#L220) of `nwbextractors.py` 
only reads the electrode properties named `rel_x`, `rel_y`, `rel_z` for determining channel locations. My NWB files were using `x`, `y`, `z`, which also follows the specification in [here](https://pynwb.readthedocs.io/en/stable/tutorials/domain/ecephys.html#electrodes-table). It did work after I change the names of those fields. This sounds like a frivolous issue. Do you think it is worth it to make this adjustment?

Thanks!

alejoe91, 2023-01-26:
Hi @oaaij-gnahz 

`rel_x/y/z` are the correct fields to store the relative location of the electrode with respect to the probe. `x/y/z`, for my understanding, would be absolute locations.

There are two options here:
1. we allow to load `x/y/z` if  `rel_x/y/z` are not set
2. we load `x/y/z` externally and set the locations
 
@CodyCBakerPhD @bendichter what do you guys think?

oaaij-gnahz, 2023-01-26:
Thanks for taking this into consideration!

h-mayorquin, 2023-01-30:
> Hi,
> 
> I was using the `NwbRecordingExtractor` and the channel locations could not be read.
> 
> From [Line 220-232](https://github.com/SpikeInterface/spikeinterface/blob/master/spikeinterface/extractors/nwbextractors.py#L220) of `nwbextractors.py` it seems to only read the electrode properties named `rel_x`, `rel_y`, `rel_z` for determining channel locations. My NWB files were using `x`, `y`, `z`, which also follows the specification in [here](https://pynwb.readthedocs.io/en/stable/tutorials/domain/ecephys.html#electrodes-table). It did work after I change the names of those fields. This sounds like a frivolous issue. Do you think it is worth it to make this adjustment?
> 
> Thanks!

@oaaij-gnahz Are you locations absolute or relative?

I think the documentation in the link is underspecified.
See:
https://pynwb.readthedocs.io/en/stable/pynwb.file.html#pynwb.file.NWBFile.add_electrode

Does it makes sense to load absolute locations (x, y and z) as the probe geometry? Is there a standard meaning for the reference `(0, 0)` in probeinterface, @alejoe91?

oaaij-gnahz, 2023-01-30:
> @oaaij-gnahz Are you locations absolute or relative?
>
> I think the documentation in the link is underspecified. See: [...]

@h-mayorquin Thank you for the pointer! My locations are relative to the probe; since there is only one `ElectrodeGroup` in the probe I am using, the relevant and absolute locations do not matter to me. After checking the link you shared, it probably does make more sense to use relative locations when converting to NWB from raw recording, at least for my scenario.

h-mayorquin, 2023-01-31:
@oaaij-gnahz 
Good that it makes sense.

We work closely with the developers of pynwb.  You have a suggestion on we could make this disctinction (position in x and relative in rel_x) more salient? 

The question remains of whether we should add function in the NWBExtractor to load the `x`,`y` and `z` as channel_locations when `rel_x`, `rel_y` and `rel_z` are not available. For probe geometry purposes what matters is the distances between them so absolute vs relative should not matter. What do you think @alejoe91?

alejoe91, 2023-01-31:
I would tend to stick to the "official" `rel_x/y/z` at this point.

@oaaij-gnahz you can always load `x/y` externally and set locations:

```
x = nwbfile.electrodes["x"][:]
y = nwbfile.electrodes["y"][:]

locations = np.zeros((len(x), 2))
locations[:, 0] = x
locations[:, 1] = y

recording.set_channel_locations(locations)
```

closed 2023-02-01 by oaaij-gnahz