SpikeInterface/spikeinterface #1217: Allow running metrics from waveform extractor in recording-less mode

TomBugnon, 2023-01-24:
Hi Alessio, Samuel and all,

I'm opening this issue to track progress regarding the possibility of running quality metrics from the waveform extractor alone (in mode "with_recording=False")

The rationale is that operations on the sorter output should be as self-contained as possible in general (decoupled from the raw recordings). Metrics might need be recomputed for specific analysis or to keep track with new metrics, and keeping the processed bin files alongside the sorter output (in order to be able to instantiate a recording) is in general too costly in terms of disk space, while relying on the raw data hinders data sharing and would require re-processing the whole recording every time we need to do some postprocessing.  

I see that currently, even `isi_violations` and  `compute_firing_rate` require the recording (only to pull the sampling frequency and total duration). `compute_snr` relies on the noise levels which could also in theory be obtained from the early samples of waveforms.

Amplitudes, locations, PCA components could ideally be computed only on the subset of extracted waveforms.

Best,
Tom

alejoe91, 2023-01-24:
Hi @TomBugnon 

this is a great point! In order to properly do that, we should:
1. extend the [`rec_attributes`](https://github.com/SpikeInterface/spikeinterface/blob/master/spikeinterface/core/waveform_extractor.py#L72), including all recording properties and segment durations (the sampling_frequency can be access through the `we.sorting`).  I think we should also cache the `recording._properties` dictionary and make a `get_channel_property()` (and `get_unit_property()`) at the waveform extractor level. This would take care of `isi_violations` and `compute_firing_rate`.
2. the `compute_noise_levels` should be a `WaveformExtension`, so that the noise levels are cached to disk (and can be reloaded) - take care of `snr`
3. I think that amplitudes, locations, and PCA components should be computed when the recording is available (than all metrics could be recomputed). This would still decouple the `qualitymetrics` from the recording, but not the `postprocessing` module.

Do you have time/bandwidth to work on this? We can have a chat the plan discuss more!

Cheers
Alessio

samuelgarcia, 2023-01-24:
Also importantly if you still have the raw recording on your path and but only deleted the preprocesed reccording binar file, you can make a WaveformExtractor on top of the lazy recording, it will be bit slower but more flexible.

PR #1258 (2023-02-03): Extend WaveformExtractor `rec_attributes` and propagate recordingless mode
fixes #1217

This PR extends the `rec_attributes` adding the following: `num_samples`, `is_filtered`, `properties` (excluding `contact_vector`).

The completely be independent of the underlying recording, the `WaveformExtractor` now has the following methods:
- `get_num_samples()`
- `get_total_samples()`
- `get_total_duration()`
- `is_filtered()`
- `get_recording_property()` (and `get_sorting_property`)

This allwo most modules to be compatible with recordingless modes:
- `curation` (which required `MockWavevorm` to ext [... cut]
closed 2023-02-03 by samuelgarcia