SpikeInterface/spikeinterface #1218: Combinato fails due to path not found

ErnestJohnston, 2023-01-24:
I am trying to run SpikeInterface using Combinato on Windows 10, but SpikeInterface cannot find the Combinato path. Regardless of whether I use `ss.run_combinato()` or `ss.run_sorter(sorter_name = 'combinato')`, I get the following error message:

`"message": "The sorter combinato is not installed.Please install it with:  \n\nTo use Combinato run:\n\n        >>> git clone https://github.com/jniediek/combinato\n    Then inside that folder, run :\n\n    >>> python3 setup_options.py\n    Finally provide the installation path by setting the COMBINATO_PATH\n    environment variables or using CombinatoSorter.set_combinato_path().\n\n\n\n    More information on Combinato at:\n        https://github.com/jniediek/combinato/wiki\n     "`

I have tried both methods to set the Combinato path that the SpikeInterface documentation recommends:

- I created a new user variable called "COMBINATO_PATH" with the value set to my combinato folder path (`'C:/Users/EJ/combinato'`). When I test this by running `os.environ['COMBINATO_PATH']` in my Python Notebook (.ipynb), it returns the correct path. 
- I set the Combinato path in my Python Notebook using `si.sorters.CombinatoSorter.set_combinato_path('C:/Users/EJ/combinato')`.

I also tested my Combinato installation (using the same data except in .h5 format) to confirm that it is working when used as a stand-alone spike sorter.

Any help is greatly appreciated!

alejoe91, 2023-01-24:
Hi, did you run the `python setup_options.py` from the combinato folder ? This is the only issue I can think of!

ErnestJohnston, 2023-01-24:
> Hi, did you run the `python setup_options.py` from the combinato folder ? This is the only issue I can think of!

Yep, for sure! That's also necessary to get Combinato working as a stand-alone spike sorter.

alejoe91, 2023-01-24:
So this is the function that check if it's installed: https://github.com/SpikeInterface/spikeinterface/blob/master/spikeinterface/sorters/combinato/combinato.py#L23

Can you try to copy/paste it in a script/notebook and run it? Maybe there's something wrong in the path handling

ErnestJohnston, 2023-01-26:
> So this is the function that check if it's installed: https://github.com/SpikeInterface/spikeinterface/blob/master/spikeinterface/sorters/combinato/combinato.py#L23
> 
> Can you try to copy/paste it in a script/notebook and run it? Maybe there's something wrong in the path handling

Okay I realized that I had appended `.py` to all of the css files in the Combinato installation in order for it to work as a stand-alone spike sorter. Removing these file extensions allowed me to initiate Combinato from SpikeInterface.

However, Combinato still fails with an assertion error because it eventually calls the `initialize_folder()` method in `spikeinterface/sorters/basesorter.py`. The `.dump_to_json()` call on Line 133 inside the `initialize_folder()` method causes the problem:

    #  if recording.is_dumpable:
    #       recording.dump_to_json(rec_file)
    #  else:
    #       d = {'warning': 'The recording is not dumpable'}
    #       rec_file.write_text(json.dumps(d, indent=4), encoding='utf8')`

My recording is a NumpyRecording object, which has the `.is_dumpable` property set to True by default. So it will run `recording.dump_to_json(rec_file)` -- but the `.dump_to_json()` method has a second check called `_check_if_dumpable()`, which returns False for the NumpyRecording object.

I tried to get around this by manually setting `recording.is_dumpable = False` before initiating Combinato. But I'm getting more errors that I am still debugging.

closed 2023-01-26 by ErnestJohnston

alejoe91, 2023-01-27:
@ErnestJohnston the `NumpyRecording` is an in-memory object and it is mainly designed for testing. While it's handy in some cases, I always recommend to save it to a binary file before further processing:
```
recording = recording.save(folder='some-folder')
```
Now the recording is persistent on disk, and this makes it suitable for parallel processing (which is tricky for memory objects and it's therefore not supported)