SpikeInterface/spikeinterface #1223: Add 'extra requirement' for quality metric sub-module

CodyCBakerPhD, 2023-01-27:
Issue discovered by adjusting dependencies on https://github.com/catalystneuro/neuroconv/pull/234, an extractor used in the testing suite uses some methods from the quality metrics submodule, and has minimal spikeinterface + scikit-learn installed, but ultimately results in some obscure errors about some level of `compute_quality_metrics` returning `None` instead of an object with a `_fields` attribute

The same code was previously working without issue, the only thing changed on that PR is the order and extend of secondary dependencies so I suspect that's the source of the problem

A simple fix would be to add an additional 'extra_requires' category for using the QM module, since I don't expect to need everything in `[full]` (or do I?)

alejoe91, 2023-01-30:
@CodyCBakerPhD that's a good idea. We could definitely make module specific extra requirements!

CodyCBakerPhD, 2023-03-07:
Uhh I might be misusing it on the NeuroConv testing suite, granted, but [according to that](https://github.com/catalystneuro/neuroconv/blob/main/requirements-testing.txt#L5-L6) it's

```
scikit-learn  # For SI Waveform tests
numba; python_version <= '3.10'  # For SI Waveform tests
```

when otherwise only using minimal SI (not `full`) installation

PR #2135 (2023-10-30): Add quality_metrics specific module requirements to `pyproject.toml`
Fixes #1223 

There could always be more submodule specific options in the toml, but since qm is a smaller module and since it was requested should be an easy addition.

Also deleted a couple of unused imports.
closed 2023-10-30 by samuelgarcia