SpikeInterface/spikeinterface #1225: Unable to run processes with multiple jobs

joaohnp, 2023-01-27:
Hi, 

So I've been trying spikeinterface==0.96.1, mostly sorting with tridesclous. 
I'm using tetrode data and recordings are somewhat long (~6h). 

I have successfully sorted parts of my (tetrodes) data, however, when I try to increase the number of jobs it crashes. Even if I slice 5min of data and try to use 2 jobs. 

If I leave it with n_jobs = 1 it does get to the end, but it's not very practical as it gets to be too lengthy. 

I get the error: PermissionError: [WinError 32] The process cannot access the file because it is being used by another process

From reading other issues I could see this being reported before, but the solutions others have tried did not solve my case. 
Thanks a lot for your help!

alejoe91, 2023-01-27:
Hi @joaohnp 

Can you post the entire error?

joaohnp, 2023-01-27:
Sure! here it goes

   File "C:\Users\acquisition\anaconda3\envs\spikeinterface\lib\multiprocessing\spawn.py", line 116, in spawn_main
     exitcode = _main(fd, parent_sentinel)
  File "C:\Users\acquisition\anaconda3\envs\spikeinterface\lib\multiprocessing\spawn.py", line 125, in _main
    prepare(preparation_data)
  File "C:\Users\acquisition\anaconda3\envs\spikeinterface\lib\multiprocessing\spawn.py", line 236, in prepare
    _fixup_main_from_path(data['init_main_from_path'])
  File "C:\Users\acquisition\anaconda3\envs\spikeinterface\lib\multiprocessing\spawn.py", line 287, in _fixup_main_from_path
    main_content = runpy.run_path(main_path,
  File "C:\Users\acquisition\anaconda3\envs\spikeinterface\lib\runpy.py", line 265, in run_path
    return _run_module_code(code, init_globals, run_name,
  File "C:\Users\acquisition\anaconda3\envs\spikeinterface\lib\runpy.py", line 97, in _run_module_code
    _run_code(code, mod_globals, init_globals,
  File "C:\Users\acquisition\anaconda3\envs\spikeinterface\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "c:\users\acquisition\documents\spikeinterface_tdc.py", line 67, in <module>
    sorting_KS25 = ss.run_sorter('tridesclous', recording_sub,
  File "C:\Users\acquisition\anaconda3\envs\spikeinterface\lib\site-packages\spikeinterface\sorters\runsorter.py", line 143, in run_sorter
    return run_sorter_local(**common_kwargs)
  File "C:\Users\acquisition\anaconda3\envs\spikeinterface\lib\site-packages\spikeinterface\sorters\runsorter.py", line 158, in run_sorter_local
    output_folder = SorterClass.initialize_folder(
  File "C:\Users\acquisition\anaconda3\envs\spikeinterface\lib\site-packages\spikeinterface\sorters\basesorter.py", line 123, in initialize_folder
    shutil.rmtree(str(output_folder))
  File "C:\Users\acquisition\anaconda3\envs\spikeinterface\lib\shutil.py", line 740, in rmtree
    return _rmtree_unsafe(path, onerror)
  File "C:\Users\acquisition\anaconda3\envs\spikeinterface\lib\shutil.py", line 618, in _rmtree_unsafe
    onerror(os.unlink, fullname, sys.exc_info())
  File "C:\Users\acquisition\anaconda3\envs\spikeinterface\lib\shutil.py", line 616, in _rmtree_unsafe
    os.unlink(fullname)
PermissionError: [WinError 32] The process cannot access the file because it is being used by another process: 'F:\\Recordings\\results_tdc\\raw_signals_0.raw'
C:\Users\acquisition\anaconda3\envs\spikeinterface\lib\site-packages\paramiko\transport.py:219: CryptographyDeprecationWarning: Blowfish has been deprecated
  "class": algorithms.Blowfish,
C:\Users\acquisition\anaconda3\envs\spikeinterface\lib\site-packages\jupyter_client\connect.py:27: DeprecationWarning: Jupyter is migrating its paths to use standard platformdirs
given by the platformdirs library.  To remove this warning and
see the appropriate new directories, set the environment variable
`JUPYTER_PLATFORM_DIRS=1` and then run `jupyter --paths`.
The use of platformdirs will be the default in `jupyter_core` v6
  from jupyter_core.paths import jupyter_data_dir
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\Users\acquisition\anaconda3\envs\spikeinterface\lib\multiprocessing\spawn.py", line 116, in spawn_main
    exitcode = _main(fd, parent_sentinel)
  File "C:\Users\acquisition\anaconda3\envs\spikeinterface\lib\multiprocessing\spawn.py", line 125, in _main
    prepare(preparation_data)
  File "C:\Users\acquisition\anaconda3\envs\spikeinterface\lib\multiprocessing\spawn.py", line 236, in prepare
    _fixup_main_from_path(data['init_main_from_path'])
  File "C:\Users\acquisition\anaconda3\envs\spikeinterface\lib\multiprocessing\spawn.py", line 287, in _fixup_main_from_path
    main_content = runpy.run_path(main_path,
  File "C:\Users\acquisition\anaconda3\envs\spikeinterface\lib\runpy.py", line 265, in run_path
    return _run_module_code(code, init_globals, run_name,
  File "C:\Users\acquisition\anaconda3\envs\spikeinterface\lib\runpy.py", line 97, in _run_module_code
    _run_code(code, mod_globals, init_globals,
  File "C:\Users\acquisition\anaconda3\envs\spikeinterface\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "c:\users\acquisition\documents\spikeinterface_tdc.py", line 67, in <module>
    sorting_KS25 = ss.run_sorter('tridesclous', recording_sub,
  File "C:\Users\acquisition\anaconda3\envs\spikeinterface\lib\site-packages\spikeinterface\sorters\runsorter.py", line 143, in run_sorter
    return run_sorter_local(**common_kwargs)
  File "C:\Users\acquisition\anaconda3\envs\spikeinterface\lib\site-packages\spikeinterface\sorters\runsorter.py", line 158, in run_sorter_local
    output_folder = SorterClass.initialize_folder(
  File "C:\Users\acquisition\anaconda3\envs\spikeinterface\lib\site-packages\spikeinterface\sorters\basesorter.py", line 123, in initialize_folder
    shutil.rmtree(str(output_folder))
  File "C:\Users\acquisition\anaconda3\envs\spikeinterface\lib\shutil.py", line 740, in rmtree
    return _rmtree_unsafe(path, onerror)
  File "C:\Users\acquisition\anaconda3\envs\spikeinterface\lib\shutil.py", line 618, in _rmtree_unsafe
    onerror(os.unlink, fullname, sys.exc_info())
  File "C:\Users\acquisition\anaconda3\envs\spikeinterface\lib\shutil.py", line 616, in _rmtree_unsafe
    os.unlink(fullname)
PermissionError: [WinError 32] The process cannot access the file because it is being used by another process: 'F:\\Recordings\\results_tdc\\raw_signals_0.raw'
[... 404 lines cut ...]
[... cut ...]
  File "C:\Users\acquisition\anaconda3\envs\spikeinterface\lib\concurrent\futures\_base.py", line 619, in result_iterator
    yield fs.pop().result()

  File "C:\Users\acquisition\anaconda3\envs\spikeinterface\lib\concurrent\futures\_base.py", line 444, in result
    return self.__get_result()

  File "C:\Users\acquisition\anaconda3\envs\spikeinterface\lib\concurrent\futures\_base.py", line 389, in __get_result
    raise self._exception

BrokenProcessPool: A process in the process pool was terminated abruptly while the future was running or pending.

alejoe91, 2023-01-27:
Can you try to delete the output folder yourself before running instead of letting the sorter overwite it? Indeed accessing and deleting folders with multiple jobs on windows is super painful...

joaohnp, 2023-01-27:
yes, this error was already after doing so. After deleting and restarting the environment

alejoe91, 2023-01-27:
I see, @samuelgarcia any idea?

alejoe91, 2023-01-28:
@joaohnp have you considered using docker? You just need a few extra install steps (see [here](https://spikeinterface.readthedocs.io/en/latest/containerized_sorters.html) ) and the you can just add the `docker_image=True` argument to the run function. This internally will run the spike sorting in a container (ubuntu-based), so the problem should not occur anymore.

Give it a try and let us know if you find any issues.
Cheers
Alessio

joaohnp, 2023-02-06:
@alejoe91 Turns out using docker with windows is not 100% smooth, had some trouble installing and some errors that other also had and were not solved so far. However, I moved everything to Ubuntu and it runs smoothly. I guess this issue can be closed and, if anything, maybe worth including a guideline suggesting to use Ubuntu instead of other OS (or maybe a VM? if that'd work). Thanks for the quick replies!

closed 2023-02-06 by joaohnp