SpikeInterface/spikeinterface #1226: IndexError: list index out of range

vncntprvst, 2023-01-28:
Hi,
I want to sort a multi-segment recording, but I can't find how to extract  or concatenate the segments properly. 
If I concatenate a recording with 3 segments with `recording_concat = si.concatenate_recordings([recording])` 
Then calling
```
sorting_KS25 = ss.run_kilosort2_5(recording_concat, 
                            output_folder=base_folder / 'results_KS2_5',
                            verbose=True, docker_image=True, 
                            **sorter_params)
```
results in this error: 
```
Starting container
Installing spikeinterface from sources in spikeinterface/kilosort2_5-compiled-base
Installing dev spikeinterface from remote repository
Installing extra requirements: ['neo']
Running kilosort2_5 sorter inside spikeinterface/kilosort2_5-compiled-base
Stopping container
---------------------------------------------------------------------------
SpikeSortingError                         Traceback (most recent call last)
Cell In[15], line 2
      1 # run spike sorting on entire recording
----> 2 sorting_KS25 = ss.run_kilosort2_5(recording_concat, 
      3                             output_folder=base_folder / 'results_KS2_5',
      4                             verbose=True, docker_image=True, 
      5                             **sorter_params)
      6 # Note: can't run with singularity_image on Windows because of import pwd error. Use docker_image instead.

File ~\miniconda3\envs\si_env\lib\site-packages\spikeinterface\sorters\runsorter.py:684, in run_kilosort2_5(*args, **kwargs)
    683 def run_kilosort2_5(*args, **kwargs):
--> 684     return run_sorter('kilosort2_5', *args, **kwargs)

File ~\miniconda3\envs\si_env\lib\site-packages\spikeinterface\sorters\runsorter.py:137, in run_sorter(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, docker_image, singularity_image, with_output, **sorter_params)
    135         else:
    136             container_image = singularity_image
--> 137     return run_sorter_container(
    138         container_image=container_image,
    139         mode=mode,
    140         **common_kwargs,
    141     )
    143 return run_sorter_local(**common_kwargs)

File ~\miniconda3\envs\si_env\lib\site-packages\spikeinterface\sorters\runsorter.py:578, in run_sorter_container(sorter_name, recording, mode, container_image, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, with_output, extra_requirements, **sorter_params)
    576 if run_error:
    577     if raise_error:
--> 578         raise SpikeSortingError(
    579             f"Spike sorting in {mode} failed with the following error:\n{run_sorter_output}")
    580 else:
    581     if with_output:

SpikeSortingError: Spike sorting in docker failed with the following error:
b'Traceback (most recent call last):\n  File "/Vincent/Data/sc012/sc012_0120/sc012_0120_001/in_container_sorter_script.py", line 9, in <module>\n    recording = load_extractor(\'/Vincent/Data/sc012/sc012_0120/sc012_0120_001/in_container_recording.json\')\n  File "/home/miniconda3/lib/python3.8/site-packages/spikeinterface/core/base.py", line 911, in load_extractor\n    return BaseExtractor.load(file_or_folder_or_dict, base_folder=base_folder)\n  File "/home/miniconda3/lib/python3.8/site-packages/spikeinterface/core/base.py", line 536, in load\n    extractor = BaseExtractor.from_dict(d, base_folder=base_folder)\n  File "/home/miniconda3/lib/python3.8/site-packages/spikeinterface/core/base.py", line 371, in from_dict\n    extractor = _load_extractor_from_dict(d)\n  File "/home/miniconda3/lib/python3.8/site-packages/spikeinterface/core/base.py", line 843, in _load_extractor_from_dict\n    kwargs[k] = [_load_extractor_from_dict(e) for e in v]\n  File "/home/miniconda3/lib/python3.8/site-packages/spikeinterface/core/base.py", line 843, in <listcomp>\n    kwargs[k] = [_load_extractor_from_dict(e) for e in v]\n  File "/home/miniconda3/lib/python3.8/site-packages/spikeinterface/core/base.py", line 854, in _load_extractor_from_dict\n    extractor = cls(**kwargs)\n  File "/home/miniconda3/lib/python3.8/site-packages/spikeinterface/extractors/neoextractors/openephys.py", line 114, in __init__\n    NeoBaseRecordingExtractor.__init__(self, stream_id=stream_id,\n  File "/home/miniconda3/lib/python3.8/site-packages/spikeinterface/extractors/neoextractors/neobaseextractor.py", line 43, in __init__\n    _NeoBaseExtractor.__init__(self, block_index, **neo_kwargs)\n  File "/home/miniconda3/lib/python3.8/site-packages/spikeinterface/extractors/neoextractors/neobaseextractor.py", line 25, in __init__\n    self.neo_reader = get_reader(self.NeoRawIOClass, **neo_kwargs)\n  File "/home/miniconda3/lib/python3.8/site-packages/spikeinterface/extractors/neoextractors/neobaseextractor.py", line 15, in get_reader\n    neo_reader.parse_header()\n  File "/home/miniconda3/lib/python3.8/site-packages/neo/rawio/baserawio.py", line 179, in parse_header\n    self._parse_header()\n  File "/home/miniconda3/lib/python3.8/site-packages/neo/rawio/openephysbinaryrawio.py", line 76, in _parse_header\n    explore_folder(self.dirname, self.experiment_names)\n  File "/home/miniconda3/lib/python3.8/site-packages/neo/rawio/openephysbinaryrawio.py", line 565, in explore_folder\n    recording_node = folder_structure[list(folder_structure.keys())[0]]\nIndexError: list index out of range\n'
```
I also don't know how to extract a segment and create a recording object. 
Could you explain how to do these two operations?
Thanks

alejoe91, 2023-01-28:
Hi @vncntprvst 

Concatenation is done correctly! I think there might be a problem in correctly mapping the file paths to the container when using a concatenate recording as input. I'll check!

For splitting or other segment utilities, check out the [segmentutils.py](https://github.com/SpikeInterface/spikeinterface/blob/master/spikeinterface/core/segmentutils.py) . I know this is not super covered in the docs and it's on our Todo list to add a better section on object manipulation!

Cheers,
Alessio

vncntprvst, 2023-01-28:
Ok, good to know 😊
Thanks for the link to the segment utilities.
Regarding the error, I'm also wondering if this may have to do with [this code section](https://github.com/SpikeInterface/spikeinterface/blob/406880a42171884e4085c01671e86e3eb5c57fb3/spikeinterface/sorters/runsorter.py#L488), as the container starts by installing the repos
```
Installing dev spikeinterface from remote repository
Installing extra requirements: ['neo']
```
This issue comes up on Windows computers, by the way, I forgot to specify that.

samuelgarcia, 2023-01-30:
@vncntprvst thanks for this.
Exactly I think we do not resolve the full perents for extra installation inside the container.

I think that more globally we should refactor the API when we run sorters through a container to have a clear option like: ```container_pip_install = ["neo==0.12", ]``` so the end user can force any extra install manually to avoid faillure whan the "magic install" fails.

PR #1230 (2023-01-30): Correctly find parent folders for concatenate recording/sorting objects
Fixes #1226

@vncntprvst @samuelgarcia the problem was that the `recording_list` was not correctly accessed by the recursive path modifier used to find common parent folders. This fix simply loops through lists and checks if there are extractors there
closed 2023-01-30 by samuelgarcia