SpikeInterface/spikeinterface #1231: Problem using AxonaRecordingExtractor: incorrect number of channels

SusanL82, 2023-01-30:
Hello, 

When I use the AxonaRecordingExtractor to load my recordings, I get the incorrect number of channels. 
My recordings have 64 channels and a tracking signal, so I'd expect either 64 or 65 channels (depending on how the tracking is handled here). Instead, I seem to get only 4.  

My code so far is like this: 

```
import spikeinterface as si
import spikeinterface.extractors as se
file_path = "K:/SEPSIS/RAW/BIN/HD132-1008_05.bin" 
ThisRec = se.AxonaRecordingExtractor(file_path)
```
If I then try to get some recording information with ThisRec.get_channel_ids I get "BaseRecording.get_channel_ids of AxonaRecordingExtractor: 4 channels - 1 segments - 24.0kHz - 1084.513s" 
The sampling rate and recording duration are correct, but the number of channels is not. Is there a way to make this extractor read all my channels? 

I have tried to change the all_annotations option in the AxonaRecordingExtractor, but adding anything beyond the file_path variable gives me an error. 

I've uploaded one of my axona .bin and .set files here, in case that helps (it's about 3.5GB): https://owncloud.cesnet.cz/index.php/s/S73ztHQrLINk0wI

alejoe91, 2023-02-01:
@SusanL82 I looked into this and 4 channels seems to be correct in this case (at least using the methods that we use to infer the number of channels).

This [function in neo](https://github.com/NeuralEnsemble/python-neo/blob/09ecccdaf84f29054ee937bc795ee55f32eeee6d/neo/rawio/axonarawio.py#L481-L495) checks the number of active tetrodes by probing the `.set` file for the key `collectMask_*`, where * is the tetrode_id. If the value of this key is '1', the status of the tetrode is active and we count 4 channels.

Are you able to access all 64 channels using the Axona tools? Maybe there is something wrong in the set file (or our way of reading the file is not correct).

Let us know what you think!

SusanL82, 2023-02-10:
I think there is indeed something up with my .set file. When I change the collectMask_* settings I seem to get the correct number of signals. I have no idea why we have a different number listed in the .set file and in the .bin file. 

Is it also possible to extract the position tracking information with the AxonaRecordingExtractor? I wonder if I should change anything in my .set file to obtain these data as well.

alejoe91, 2023-02-10:
I see, good to know that changing the collect mask seems to be working.

We don't have a way to load the tracking signals, since SpikeInterface is only for ephys data. 

You might find smething interesting in these other repos:

- https://github.com/catalystneuro/hussaini-lab-to-nwb/tree/main/hussaini_lab_to_nwb
- https://github.com/CINPLA/pyxona (might be a bit outdated)

alejoe91, 2023-02-22:
@SusanL82 can we close the issue? Feel free to open a new one if you have other questions/problems

SusanL82, 2023-03-01:
Yes. My problem is basically solved when I edit the .set files. 
Thanks for your help!