SpikeInterface/spikeinterface #1234: Save function using up all the memory

ziyuw2, 2023-01-30:
Hi there,
I don't know if this has been brought to your attention before, but the save function seemed to use up all the memory, resulted in my pc crashing three times (16gb of ram) when I ran this function, and my lab mate @vncntprvst 's pc crashing once (32gb of ram). Is there a way for us to allocate only part of our memory for this function? Please let me know! Thank you for your time!

`if (base_folder / "preprocessed").is_dir():
    recording_saved = si.load_extractor(base_folder / "preprocessed")
else:
    recording_saved = recording_all_cmr.save(folder=base_folder / "preprocessed", **job_kwargs)`

vncntprvst, 2023-01-31:
Just adding that we tried saving using chunks, with something like `recording_all.save(format='binary', folder=r'C:\data_folder', n_jobs=10, chunk_duration='1s', progress_bar=True)`, but that did not alleviate the problem.

alejoe91, 2023-01-31:
Hi, the function should only use roughly 1s * number of jobs worth of memory..on Linux I'm 100% that is the case :) is this on windows? Can you give us some specs on the `recording_all` object?

samuelgarcia, 2023-01-31:
On windows it is more complicated because if the underlying format reader use memmap (which is the case for many format) then the chunk memory are released only if necessary and as soon there  enought ram available then memory chunks are NOT released until the memory is full.
So sometimes you can expect a high memory usage but normally no crash.

yger, 2023-01-31:
Actually, I think I encountered similar problems a while ago with spyking-circus and memmap on windows. The solution was to force the file to be open/close after each chunk is read, to ensure memory is flushed. This is not slower, and it prevents any memory leak

vncntprvst, 2023-01-31:
Thanks Pierre. It's great to know because for calcium imaging analysis we're likely to make greater use of memmaps, so this tip might help us across the board :)

samuelgarcia, 2023-01-31:
@yger : I know but this would make an enormous change in many reader in neo.

vncntprvst, 2023-02-01:
We may switch some processing to WSL or Linux platforms. In the meantime, as a workaround, is there a format that does not use memmap that we could convert recordings to before saving?

alejoe91, 2023-02-01:
You can try to use `zarr` (and you can even compress the data):

```
recording_saved = recording_all_cmr.save(folder=base_folder / "preprocessed", format='zarr', **job_kwargs)
```

I don't think that's a long-term solution though..running most of the sorters will require to re-save to binary..

h-mayorquin, 2023-03-07:
@ziyuw2 Did using zarr worked for you? Also, what was the error that you were getting more specifically?

vncntprvst, 2023-03-08:
@h-mayorquin We didn't try using zarr at the time (@ziyuw2 let us know if you did try). We just avoided saving the recording when running this code on Windows. There was no specific error, just the memory filling up until everything slowed down, and often crashing the machine. 
I can give zarr a try tomorrow.

vncntprvst, 2023-03-30:
Using zarr, the memory usage still gets dangerously close to 100%. See screenshot below. So far no crash when saving data, though. Perhaps zarr is better at avoiding a full system face-plant.  

[image]

samuelgarcia, 2023-03-31:
If the zarr writting instead of emmap writting do not change the behavior, then it should be on the reading side. And so the neo memmap reading.
I promise I will try to make some tests.

vncntprvst, 2023-04-03:
On that topic, I'm still unclear how one loads a "zarr" folder. The [documentation says](https://spikeinterface.readthedocs.io/en/latest/modules/core.html#saving-loading-and-compression): the saved object can be reloaded with the `load_extractor()` function.
But doing `recording_saved = si.load_extractor(base_folder / "preprocessed.zarr")` results in the "This folder is not a cached folder" error:
```
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
Cell In[19], line 1
----> 1 recording_saved = si.load_extractor(base_folder / "preprocessed.zarr")
      2 print(recording_saved)

File ~/mambaforge/envs/si_env/lib/python3.9/site-packages/spikeinterface/core/base.py:911, in load_extractor(file_or_folder_or_dict, base_folder)
    909     return BaseExtractor.from_dict(file_or_folder_or_dict, base_folder=base_folder)
    910 else:
--> 911     return BaseExtractor.load(file_or_folder_or_dict, base_folder=base_folder)

File ~/mambaforge/envs/si_env/lib/python3.9/site-packages/spikeinterface/core/base.py:557, in BaseExtractor.load(file_path, base_folder)
    554     file = f
    556 if file is None:
--> 557     raise ValueError(f'This folder is not a cached folder {file_path}')
    558 extractor = BaseExtractor.load(file, base_folder=folder)
    560 return extractor

ValueError: This folder is not a cached folder /home/wanglab/data/whisker_asym/sc014/sc014_0324/sc014_0324_001/preprocessed.zarr
```

alejoe91, 2023-04-03:
Oups we have to update the docs. Currently you have to use the read_zarr function and just provide the zarr path

alejoe91, 2023-04-03:
But I think it makes sense to have the load_extractor also support the zarr format directly

vncntprvst, 2023-04-03:
I think so too :)
I have a different issue now, that I'll post separately. 
Thanks!

h-mayorquin, 2023-06-05:
This should be solved by https://github.com/SpikeInterface/spikeinterface/pull/1602.

h-mayorquin, 2023-07-19:
@vncntprvst I am closing this issue as I think the changes in https://github.com/SpikeInterface/spikeinterface/pull/1602 should have solve this problem. Feel free to open this again -or preferably another issue- if you have a smiliar problems. I would like to solve the library memory problems as much as possible.

closed 2023-07-19 by h-mayorquin