SpikeInterface/spikeinterface #1235: Serialization project outline

h-mayorquin, 2023-01-31:
Hi @alejoe91 @samuelgarcia 
I want to propose here an order to work in a series of serialization issues, discuss and get your feedback and direction. I would like to work on this in the following order, each item being a tailored PR to keep the workflow modular and specific.  Let me know what you think of the order and priorities.

- [ ] Adding serialization capabilities to recorders and sorters  #1227. 
- [ ] Using the serialization capabilites to add initial instantiation capabilties to graph computation pipeline that Sam is proposing in #1209 (after the PR is merged) [I remember the problem here was serialization and I want to tackle this first as it was a blocker for some of my other work on denoising].
- [ ] Using the serialization capabilites to remove `to_dict` mechanism in parallel pipelines [this is a big refactor so I am doing it third]
- [ ] Using the same capabilities to avoid random and computation in preprocessing init class (solving issue #1118).
- [ ] Separate serialization and json-nification infraestructure.

What do you think? If you agree with this proposal I would like to request a review of #1227 so I can start working on the other items.

alejoe91, 2023-01-31:
Hi Ramon!

It sounds like a good plan to me and I totally endorse it. @samuelgarcia what do you think?

samuelgarcia, 2023-01-31:
Cool.
Lets discuss thsi with a call soon.
Remove the to_dict in parralel processing is good.

Here some task are independant. I would make separate small PRs for fast discussion and merging.

h-mayorquin, 2023-02-01:
Yes, I am thinking on making a PR per point above (or maybe more for the later items). Let's agree on when to do a call at the end of today's meeting : ).

h-mayorquin, 2023-07-19:
Relevant for this project:
https://github.com/SpikeInterface/spikeinterface/pull/1831

alejoe91, 2023-11-06:
@h-mayorquin can we close this? I checked the last box after #2027

The last serialization issue (https://github.com/SpikeInterface/spikeinterface/issues/1118) is kind of standalone

h-mayorquin, 2023-11-06:
Yes, I agree, and it was also wrongly scoped. I have been reading the pickling protocol section in the docs and I suspect we might be able to pass numpy buffers without serializing at all but that will be another issue!

I think the top comment looks very nice with all of those to-do checked in : D

closed 2023-11-06 by h-mayorquin