SpikeInterface/spikeinterface #1252: sigui compute_center_of_mass() got an unexpected keyword argument 'num_channels'

ziyuw2, 2023-02-01:
Hi there,
When I'm runing `!sigui waveforms/` today, I got the following error 
`C:\Users\Ziyu\anaconda3\envs\si_env\lib\site-packages\jupyter_client\connect.py:27: DeprecationWarning: Jupyter is migrating its paths to use standard platformdirs
given by the platformdirs library.  To remove this warning and
see the appropriate new directories, set the environment variable
`JUPYTER_PLATFORM_DIRS=1` and then run `jupyter --paths`.
The use of platformdirs will be the default in `jupyter_core` v6
  from jupyter_core.paths import jupyter_data_dir
C:\Users\Ziyu\anaconda3\envs\si_env\lib\site-packages\spikeinterface\postprocessing\template_tools.py:24: DeprecationWarning: The spikeinterface.postprocessing.template_tools is submodule is deprecated.Use spikeinterface.core.template_tools instead
  _warn()
C:\Users\Ziyu\anaconda3\envs\si_env\lib\site-packages\spikeinterface\postprocessing\template_tools.py:25: DeprecationWarning: The 'get_template_channel_sparsity()' function is deprecated. Use 'compute_sparsity()' instead
  return tt.get_template_channel_sparsity(*args, **kwargs)
C:\Users\Ziyu\anaconda3\envs\si_env\lib\site-packages\spikeinterface\postprocessing\template_tools.py:19: DeprecationWarning: The spikeinterface.postprocessing.template_tools is submodule is deprecated.Use spikeinterface.core.template_tools instead
  _warn()
Traceback (most recent call last):
  File "C:\Users\Ziyu\anaconda3\envs\si_env\lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\Ziyu\anaconda3\envs\si_env\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\Ziyu\anaconda3\envs\si_env\Scripts\sigui.exe\__main__.py", line 7, in <module>
  File "C:\Users\Ziyu\anaconda3\envs\si_env\lib\site-packages\spikeinterface_gui\main.py", line 34, in run_mainwindow_cli
    run_mainwindow(waveform_folder)
  File "C:\Users\Ziyu\anaconda3\envs\si_env\lib\site-packages\spikeinterface_gui\main.py", line 15, in run_mainwindow
    win = MainWindow(we)
  File "C:\Users\Ziyu\anaconda3\envs\si_env\lib\site-packages\spikeinterface_gui\mainwindow.py", line 23, in __init__
    self.controller = SpikeinterfaceController(waveform_extractor, verbose=verbose)
  File "C:\Users\Ziyu\anaconda3\envs\si_env\lib\site-packages\spikeinterface_gui\controller.py", line 132, in __init__
    self.unit_positions = compute_unit_locations(self.we, method='center_of_mass',  num_channels=10)
  File "C:\Users\Ziyu\anaconda3\envs\si_env\lib\site-packages\spikeinterface\postprocessing\unit_localization.py", line 127, in compute_unit_locations
    ulc.run()
  File "C:\Users\Ziyu\anaconda3\envs\si_env\lib\site-packages\spikeinterface\core\waveform_extractor.py", line 1629, in run
    self._run(**kwargs)
  File "C:\Users\Ziyu\anaconda3\envs\si_env\lib\site-packages\spikeinterface\postprocessing\unit_localization.py", line 60, in _run
    unit_location = compute_center_of_mass(self.waveform_extractor,  **method_kwargs)
TypeError: compute_center_of_mass() got an unexpected keyword argument 'num_channels'`

Is there any idea why this is happening or anything I can do on my end to fix this? Thank you for your time!

samuelgarcia, 2023-02-01:
Hi you should update bith spikeinterface and spikeinterface-gui from the source code.
We made some strong refactoring since some month.
The next release should happen soon (20days).

alejoe91, 2023-11-06:
I think we can close :) Working on our side too!

closed 2023-11-06 by alejoe91