SpikeInterface/spikeinterface #1253: Singularity image not found or generated

vncntprvst, 2023-02-01:
I'm trying to run SpikeInterface with containerized sorters on a computing cluster, and the code doesn't seem to find the image. 
The cluster uses Singularity, which works fine for other pipelines. 
Here, when I call 
```
sorting_KS25 = ss.run_kilosort2_5(recording_saved, 
                            output_folder=base_folder / 'results_KS2_5',
                            verbose=True, singularity_image=True, 
                            **sorter_params)
```
I get this error (full stack blow): `DockerException: Error while fetching server API version: ('Connection aborted.', FileNotFoundError(2, 'No such file or directory'))`.   
`spython` is installed and the Singularity module loaded. 
Just in case, I generated the sif file manually and copied it to `$HOME/.singularity/cache` and couple of other places, but the error still occurs. 
Specifically, it looks like the image isn't found in this section of `runsorter.py`: 
```
if Path(container_image).exists():
                singularity_image = container_image
            elif Path(sif_file).exists():
                singularity_image = sif_file
```
I don't know why it does not downloaded itself either. 
It might be connection issue (I'm connected to the HPCCfrom my laptop through VPN on a wifi), but I don't know why that would interfere with code execution.
Any idea? 

Full error stack: 
```
---------------------------------------------------------------------------
FileNotFoundError                         Traceback (most recent call last)
File /om2/user/prevosto/anaconda/envs/si_env/lib/python3.9/site-packages/urllib3/connectionpool.py:703, in HTTPConnectionPool.urlopen(self, method, url, body, headers, retries, redirect, assert_same_host, timeout, pool_timeout, release_conn, chunked, body_pos, **response_kw)
    702 # Make the request on the httplib connection object.
--> 703 httplib_response = self._make_request(
    704     conn,
    705     method,
    706     url,
    707     timeout=timeout_obj,
    708     body=body,
    709     headers=headers,
    710     chunked=chunked,
    711 )
    713 # If we're going to release the connection in ``finally:``, then
    714 # the response doesn't need to know about the connection. Otherwise
    715 # it will also try to release it and we'll have a double-release
    716 # mess.

File /om2/user/prevosto/anaconda/envs/si_env/lib/python3.9/site-packages/urllib3/connectionpool.py:398, in HTTPConnectionPool._make_request(self, conn, method, url, timeout, chunked, **httplib_request_kw)
    397     else:
--> 398         conn.request(method, url, **httplib_request_kw)
    400 # We are swallowing BrokenPipeError (errno.EPIPE) since the server is
    401 # legitimately able to close the connection after sending a valid response.
    402 # With this behaviour, the received response is still readable.

File /om2/user/prevosto/anaconda/envs/si_env/lib/python3.9/http/client.py:1285, in HTTPConnection.request(self, method, url, body, headers, encode_chunked)
   1284 """Send a complete request to the server."""
-> 1285 self._send_request(method, url, body, headers, encode_chunked)

File /om2/user/prevosto/anaconda/envs/si_env/lib/python3.9/http/client.py:1331, in HTTPConnection._send_request(self, method, url, body, headers, encode_chunked)
   1330     body = _encode(body, 'body')
-> 1331 self.endheaders(body, encode_chunked=encode_chunked)

File /om2/user/prevosto/anaconda/envs/si_env/lib/python3.9/http/client.py:1280, in HTTPConnection.endheaders(self, message_body, encode_chunked)
   1279     raise CannotSendHeader()
-> 1280 self._send_output(message_body, encode_chunked=encode_chunked)

File /om2/user/prevosto/anaconda/envs/si_env/lib/python3.9/http/client.py:1040, in HTTPConnection._send_output(self, message_body, encode_chunked)
   1039 del self._buffer[:]
-> 1040 self.send(msg)
   1042 if message_body is not None:
   1043 
   1044     # create a consistent interface to message_body

File /om2/user/prevosto/anaconda/envs/si_env/lib/python3.9/http/client.py:980, in HTTPConnection.send(self, data)
[... 203 lines cut ...]
---> 45     self.api = APIClient(*args, **kwargs)

File /om2/user/prevosto/anaconda/envs/si_env/lib/python3.9/site-packages/docker/api/client.py:197, in APIClient.__init__(self, base_url, version, timeout, tls, user_agent, num_pools, credstore_env, use_ssh_client, max_pool_size)
    192 # version detection needs to be after unix adapter mounting
    193 if version is None or (isinstance(
    194                         version,
    195                         str
    196                         ) and version.lower() == 'auto'):
--> 197     self._version = self._retrieve_server_version()
    198 else:
    199     self._version = version

File /om2/user/prevosto/anaconda/envs/si_env/lib/python3.9/site-packages/docker/api/client.py:221, in APIClient._retrieve_server_version(self)
    216     raise DockerException(
    217         'Invalid response from docker daemon: key "ApiVersion"'
    218         ' is missing.'
    219     )
    220 except Exception as e:
--> 221     raise DockerException(
    222         f'Error while fetching server API version: {e}'
    223     )

DockerException: Error while fetching server API version: ('Connection aborted.', FileNotFoundError(2, 'No such file or directory'))

```

samuelgarcia, 2023-02-02:
Hi Vincent,
this looks anoying and very hard to track.
Did you try after downloading the image  to do `singularity_image='/path/to/the/image.siff`

alejoe91, 2023-02-02:
When you use `singularity_image=True`, internally we use the docker package to fetch the correct image. This probably requires docker to be correctly setup. 
I would try, instead, to point to the correct image on dockerhub by name, something like `singularity_image="spikeinterface/kilosort25-compiled-base:latest"` or even to point to the exact version.

vncntprvst, 2023-02-02:
Thanks both for the feedback. I'll test both solutions. 
I had a similar issue also yesterday with another remote server, this time calling a Docker container, that I haven't had time to explore yet. I'll get back to you.

vncntprvst, 2023-02-02:
So running the command with the dockerhub image location (e.g., `singularity_image="spikeinterface/pykilosort-base:latest"`), results in the same error as above: `
DockerException: Error while fetching server API version: ('Connection aborted.', FileNotFoundError(2, 'No such file or directory'))`.
Now, calling the local file produces a different error. e.g.: 
```
sorting_KS25 = ss.run_kilosort2_5(recording_concat, 
                        output_folder=base_folder / 'results_KS2_5',
                        verbose=True, 
                        singularity_image='$HOME/.singularity/cache/kilosort2_5-compiled-base_latest.sif')
```
returns this: 
```
Starting container
spikeinterface is already installed in $HOME/.singularity/cache/kilosort2_5-compiled-base_latest.sif
Running kilosort2_5 sorter inside $HOME/.singularity/cache/kilosort2_5-compiled-base_latest.sif
Stopping container

---------------------------------------------------------------------------
SpikeSortingError                         Traceback (most recent call last)
Cell In[18], line 2
      1 recording_concat = si.concatenate_recordings([recording_saved])
----> 2 sorting_KS25 = ss.run_kilosort2_5(recording_concat, 
      3                             output_folder=base_folder / 'results_KS2_5',
      4                             verbose=True, singularity_image='$HOME/.singularity/cache/kilosort2_5-compiled-base_latest.sif')

File /om2/user/prevosto/anaconda/envs/si_env/lib/python3.9/site-packages/spikeinterface/sorters/runsorter.py:684, in run_kilosort2_5(*args, **kwargs)
    683 def run_kilosort2_5(*args, **kwargs):
--> 684     return run_sorter('kilosort2_5', *args, **kwargs)

File /om2/user/prevosto/anaconda/envs/si_env/lib/python3.9/site-packages/spikeinterface/sorters/runsorter.py:137, in run_sorter(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, docker_image, singularity_image, with_output, **sorter_params)
    135         else:
    136             container_image = singularity_image
--> 137     return run_sorter_container(
    138         container_image=container_image,
    139         mode=mode,
    140         **common_kwargs,
    141     )
    143 return run_sorter_local(**common_kwargs)

File /om2/user/prevosto/anaconda/envs/si_env/lib/python3.9/site-packages/spikeinterface/sorters/runsorter.py:578, in run_sorter_container(sorter_name, recording, mode, container_image, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, with_output, extra_requirements, **sorter_params)
    576 if run_error:
    577     if raise_error:
--> 578         raise SpikeSortingError(
    579             f"Spike sorting in {mode} failed with the following error:\n{run_sorter_output}")
    580 else:
    581     if with_output:

SpikeSortingError: Spike sorting in singularity failed with the following error:
Error for command "exec": unknown flag: --env

Options for exec command:

      --add-caps string        a comma separated capability list to add
      --allow-setuid           allow setuid binaries in container (root only)
      --app string             set an application to run inside a container
      --apply-cgroups string   apply cgroups from file for container
                               processes (root only)
  -B, --bind strings           a user-bind path specification.  spec has
                               the format src[:dest[:opts]], where src and
                               dest are outside and inside paths.  If dest
                               is not given, it is set equal to src. 
                               Mount options ('opts') may be specified as
                               'ro' (read-only) or 'rw' (read/write, which
                               is the default). Multiple bind paths can be
                               given by a comma separated list.
  -e, --cleanenv               clean environment before running container
  -c, --contain                use minimal /dev and empty other
                               directories (e.g. /tmp and $HOME) instead
                               of sharing filesystems from your host
  -C, --containall             contain not only file systems, but also
                               PID, IPC, and environment
[... 48 lines cut ...]
                               unprivileged on recent kernels. This
                               disables some features of Singularity, for
                               example it only works with sandbox images.
      --uts                    run container in a new UTS namespace
      --vm                     enable VM support
      --vm-cpu string          Number of CPU cores to allocate to Virtual
                               Machine (implies --vm) (default "1")
      --vm-err                 enable attaching stderr from VM
      --vm-ip string           IP Address to assign for container usage.
                               Defaults to DHCP within bridge network.
                               (default "dhcp")
      --vm-ram string          Amount of RAM in MiB to allocate to Virtual
                               Machine (implies --vm) (default "1024")
  -W, --workdir string         working directory to be used for /tmp,
                               /var/tmp and $HOME (if -c/--contain was
                               also used)
  -w, --writable               by default all Singularity containers are
                               available as read only. This option makes
                               the file system accessible as read/write.
      --writable-tmpfs         makes the file system accessible as
                               read-write with non persistent data (with
                               overlay support only)

Run 'singularity exec --help' for more detailed usage information.
```

vncntprvst, 2023-02-02:
The currently installed versions:
`singularity version 3.4.1`
`python --version Python 3.9.15`
`spikeinterface` and `spikeinterface-gui` installed with pip from source: 
```
pip install "spikeinterface[full,widgets] @ git+https://github.com/SpikeInterface/spikeinterface.git"
pip install "spikeinterface-gui @ git+https://github.com/SpikeInterface/spikeinterface-gui.git"
```
`conda list`
```
# packages in environment at /om2/user/prevosto/anaconda/envs/si_env:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       2_gnu    conda-forge
aiohttp                   3.8.3                    pypi_0    pypi
aiosignal                 1.3.1                    pypi_0    pypi
alabaster                 0.7.13             pyhd8ed1ab_0    conda-forge
alsa-lib                  1.2.3.2              h166bdaf_0    conda-forge
altair                    4.2.2                    pypi_0    pypi
anyio                     3.6.2              pyhd8ed1ab_0    conda-forge
appdirs                   1.4.4              pyh9f0ad1d_0    conda-forge
argon2-cffi               21.3.0             pyhd8ed1ab_0    conda-forge
argon2-cffi-bindings      21.2.0           py39hb9d737c_3    conda-forge
asciitree                 0.3.3                    pypi_0    pypi
asttokens                 2.2.1              pyhd8ed1ab_0    conda-forge
async-timeout             4.0.2                    pypi_0    pypi
attrs                     22.2.0             pyh71513ae_0    conda-forge
babel                     2.11.0             pyhd8ed1ab_0    conda-forge
backcall                  0.2.0              pyh9f0ad1d_0    conda-forge
backports                 1.0                pyhd8ed1ab_3    conda-forge
backports.functools_lru_cache 1.6.4              pyhd8ed1ab_0    conda-forge
beautifulsoup4            4.11.1             pyha770c72_0    conda-forge
bleach                    6.0.0              pyhd8ed1ab_0    conda-forge
blosc                     1.21.3               hafa529b_0    conda-forge
brotli                    1.0.9                h166bdaf_8    conda-forge
brotli-bin                1.0.9                h166bdaf_8    conda-forge
brotlipy                  0.7.0           py39hb9d737c_1005    conda-forge
bzip2                     1.0.8                h7f98852_4    conda-forge
c-ares                    1.18.1               h7f98852_0    conda-forge
ca-certificates           2022.12.7            ha878542_0    conda-forge
cachecontrol              0.12.11                  pypi_0    pypi
cached-property           1.5.2                hd8ed1ab_1    conda-forge
cached_property           1.5.2              pyha770c72_1    conda-forge
cachetools                5.3.0                    pypi_0    pypi
cbor2                     5.4.6                    pypi_0    pypi
certifi                   2022.12.7          pyhd8ed1ab_0    conda-forge
cffi                      1.15.1           py39he91dace_3    conda-forge
charset-normalizer        2.1.1              pyhd8ed1ab_0    conda-forge
click                     8.1.3                    pypi_0    pypi
cloudpickle               2.2.1              pyhd8ed1ab_0    conda-forge
colorama                  0.4.6              pyhd8ed1ab_0    conda-forge
colorcet                  3.0.1                    pypi_0    pypi
comm                      0.1.2              pyhd8ed1ab_0    conda-forge
contourpy                 1.0.7            py39h4b4f3f3_0    conda-forge
cryptography              39.0.0           py39hd598818_0    conda-forge
cycler                    0.11.0             pyhd8ed1ab_0    conda-forge
cython                    0.29.33          py39h227be39_0    conda-forge
dask                      2023.1.1                 pypi_0    pypi
dbus                      1.13.6               h5008d03_3    conda-forge
debugpy                   1.6.6            py39h227be39_0    conda-forge
decorator                 5.1.1              pyhd8ed1ab_0    conda-forge
defusedxml                0.7.1              pyhd8ed1ab_0    conda-forge
distinctipy               1.2.2                    pypi_0    pypi
distributed               2023.1.1                 pypi_0    pypi
docker                    6.0.1                    pypi_0    pypi
docutils                  0.19             py39hf3d152e_1    conda-forge
entrypoints               0.4                pyhd8ed1ab_0    conda-forge
et-xmlfile                1.1.0                    pypi_0    pypi
executing                 1.2.0              pyhd8ed1ab_0    conda-forge
[... 254 lines cut ...]
tridesclous               1.6.6.1                  pypi_0    pypi
typing-extensions         4.4.0                hd8ed1ab_0    conda-forge
typing_extensions         4.4.0              pyha770c72_0    conda-forge
tzdata                    2022g                h191b570_0    conda-forge
unicodedata2              15.0.0           py39hb9d737c_0    conda-forge
urllib3                   1.26.14            pyhd8ed1ab_0    conda-forge
watchdog                  2.2.1                    pypi_0    pypi
wcwidth                   0.2.6              pyhd8ed1ab_0    conda-forge
webencodings              0.5.1                      py_1    conda-forge
websocket-client          1.5.0              pyhd8ed1ab_0    conda-forge
wheel                     0.38.4             pyhd8ed1ab_0    conda-forge
widgetsnbextension        4.0.5              pyhd8ed1ab_0    conda-forge
[... cut ...]
yaml                      0.2.5                h7f98852_2    conda-forge
yarl                      1.8.2                    pypi_0    pypi
zarr                      2.13.6                   pypi_0    pypi
zeromq                    4.3.4                h9c3ff4c_1    conda-forge
zict                      2.2.0                    pypi_0    pypi
zipp                      3.12.0             pyhd8ed1ab_0    conda-forge
zlib                      1.2.13               h166bdaf_4    conda-forge
zstd                      1.5.2                h3eb15da_6    conda-forge
```

vncntprvst, 2023-02-06:
I was able to fix the exact same error that occurred with Docker engine on a Linux server. The call 
```
sorting_KS25 = ss.run_kilosort2_5(recording_saved, 
                            output_folder=base_folder / 'results_KS2_5',
                            verbose=True, singularity_image=True, 
                            **sorter_params)
```
returned: `DockerException: Error while fetching server API version`

What was happening was the Docker engine was using on a different context than the default one, because it was installed as part of Docker Desktop. From the documentation: 

> When Docker Desktop starts, it creates a dedicated [context](https://docs.docker.com/engine/context/working-with-contexts) that the Docker CLI can use as a target and sets it as the current context in use. This is to avoid a clash with a local Docker Engine that may be running on the Linux host and using the default context. On shutdown, Docker Desktop resets the current context to the previous one.

The fix was simply to install [Docker engine](https://docs.docker.com/engine/install/ubuntu/) separately, so that it runs in the default context.  

Now, listing Docker contexts with `docker context ls`, returns this: 
```
NAME                TYPE                DESCRIPTION                               DOCKER ENDPOINT                                    KUBERNETES ENDPOINT   ORCHESTRATOR
default *           moby                Current DOCKER_HOST based configuration   unix:///var/run/docker.sock
desktop-linux       moby                                                          unix:///home/wanglab/.docker/desktop/docker.sock
``` 
Note that before installing Docker engine separately, `/var/run/docker.sock` did not exist, even though there was a fully functional Docker installed. 
@alejoe91 @samuelgarcia , I think this means the SI code that calls the Docker engine is not able to deal with non-default contexts. This also gives us a clue about what may not be working on that computing cluster with respect to calling Singularity containers.

vncntprvst, 2023-02-06:
I solved the issue for Singularity containers, and it was actually much simpler than what I discussed above. 
The options passed to the Singularity client are [`options = ['--cleanenv', '--env', f'PYTHONUSERBASE={self.py_user_base}']`](https://github.com/SpikeInterface/spikeinterface/blob/2068809ee7626175210b5071b801962883a95615/spikeinterface/sorters/runsorter.py#L322). 
However, the `--env` flag was only introduced in [Singularity 3.6](https://docs.sylabs.io/guides/3.7/user-guide/environment_and_metadata.html#changes-in-singularity-3-6). So to run Singularity on computing clusters, one needs to load a recent enough Singularity module (e.g., 3.6.3 - on my institute's cluster, the default version is 3.4.1). 

In addition, when running the code on a platform that does not have Docker installed, it's probably wise to not install the `docker` python package, since it may get this [bit of code](https://github.com/SpikeInterface/spikeinterface/blob/2068809ee7626175210b5071b801962883a95615/spikeinterface/sorters/runsorter.py#L270) executed.

closed 2023-02-06 by vncntprvst

alejoe91, 2023-02-06:
Thanks for tracking this down @vncntprvst !!!