SpikeInterface/spikeinterface #1256: issues with running kilosort3 container on concatenated files

LaurenzMuessig, 2023-02-02:
Hi,

In my pipeline I concatenate a few sessions (np1.0 data recorded using spikeGLX), preprocess them and then write the file to disk using save.

However, if I now call the sorter, e.g.

sorting = ss.run_kilosort3(recording=rec_Fin,output_folder=outDir,docker_image=True,**KS3Params)

I get the following error:

SpikeSortingError: Spike sorting in docker failed with the following error:
b'Traceback (most recent call last):\n  File "/media/SSD2/kilosort_LM/batchClust/in_container_sorter_script.py", line 9, in <module>\n    recording = load_extractor(\'/media/SSD2/kilosort_LM/batchClust/in_container_recording.json\')\n  File "/home/miniconda3/lib/python3.8/site-packages/spikeinterface/core/base.py", line 911, in load_extractor\n    return BaseExtractor.load(file_or_folder_or_dict, base_folder=base_folder)\n  File "/home/miniconda3/lib/python3.8/site-packages/spikeinterface/core/base.py", line 536, in load\n    extractor = BaseExtractor.from_dict(d, base_folder=base_folder)\n  File "/home/miniconda3/lib/python3.8/site-packages/spikeinterface/core/base.py", line 371, in from_dict\n    extractor = _load_extractor_from_dict(d)\n  File "/home/miniconda3/lib/python3.8/site-packages/spikeinterface/core/base.py", line 837, in _load_extractor_from_dict\n    kwargs[k] = _load_extractor_from_dict(v)\n  File "/home/miniconda3/lib/python3.8/site-packages/spikeinterface/core/base.py", line 837, in _load_extractor_from_dict\n    kwargs[k] = _load_extractor_from_dict(v)\n  File "/home/miniconda3/lib/python3.8/site-packages/spikeinterface/core/base.py", line 837, in _load_extractor_from_dict\n    kwargs[k] = _load_extractor_from_dict(v)\n  [Previous line repeated 1 more time]\n  File "/home/miniconda3/lib/python3.8/site-packages/spikeinterface/core/base.py", line 843, in _load_extractor_from_dict\n    kwargs[k] = [_load_extractor_from_dict(e) for e in v]\n  File "/home/miniconda3/lib/python3.8/site-packages/spikeinterface/core/base.py", line 843, in <listcomp>\n    kwargs[k] = [_load_extractor_from_dict(e) for e in v]\n  File "/home/miniconda3/lib/python3.8/site-packages/spikeinterface/core/base.py", line 854, in _load_extractor_from_dict\n    extractor = cls(**kwargs)\n  File "/home/miniconda3/lib/python3.8/site-packages/spikeinterface/extractors/neoextractors/spikeglx.py", line 51, in __init__\n    NeoBaseRecordingExtractor.__init__(self, stream_id=stream_id, \n  File "/home/miniconda3/lib/python3.8/site-packages/spikeinterface/extractors/neoextractors/neobaseextractor.py", line 67, in __init__\n    assert stream_id in stream_ids, f\'stream_id {stream_id} is not in {stream_ids}\'\nAssertionError: stream_id imec0.ap is not in []\n'

This seems to be caused by the fact that there isn't a .meta file for the concatenated session. If I just take a single session and write the recording to disk, I can run the sorter with no issues. Am I doing something wrong here or do I have to generate a .meta file for the concatenated session for the sorting to work? Thanks for you help.

Cheers

Laurenz

alejoe91, 2023-02-02:
Hi @LaurenzMuessig 

Can you install from master? 
I think this was recently fixed by this PR #1230

LaurenzMuessig, 2023-02-02:
Thanks a lot for your quick response @alejoe91 . That fixed it. 

Cheers

Laurenz

closed 2023-02-02 by LaurenzMuessig