SpikeInterface/spikeinterface #1260: Incorrect channel IDs in `SpikeGadgetsRecordingExtractor`

khl02007, 2023-02-02:
Recordings from SpikeGadgets hardware have two possible channel maps: HWChan (a channel map defined by the amplifier) and Trodes channel map (for displaying on Trodes, the data acquisition software for SpikeGadgets). Both of these are in the header of the file that contains the recording (in `.rec` format). The channel IDs one gets with `get_channel_ids` from a `SpikeGadgetsRecordingExtractor` is in the HWChan order, which is sensible. But the order of the traces is completely scrambled (neither Trodes nor HWChan). Seems like this recording extractor relies on `neo`. @samuelgarcia @alejoe91 Could you take a look?

alejoe91, 2023-02-11:
Hi @khl02007 

Can you post the issue on NEO and share a test data file?

khl02007, 2023-02-12:
sure, now here https://github.com/NeuralEnsemble/python-neo/issues/1215

zm711, 2024-07-24:
Closed by https://github.com/NeuralEnsemble/python-neo/pull/1496

Will require an install from source for now.

closed 2024-07-24 by zm711