SpikeInterface/spikeinterface #1311: Exceeding storage allocation for CI caches

h-mayorquin, 2023-02-13:
In some parts of the repo (right now #1310) we are getting errors like this:

[image]

I think they are generated by an error on virtual environment caching. This in turn most likely is coming from our large cache sizes which are over the quota right now: 
https://github.com/SpikeInterface/spikeinterface/actions/caches

This situation causes cache trashing as described here:
https://docs.github.com/en/actions/using-workflows/caching-dependencies-to-speed-up-workflows#usage-limits-and-eviction-policy

Importantly, note that we have repeated caches with the same hash and date for different pull requests. I think this is happening because of the cache accessing logic of github CI:
https://docs.github.com/en/actions/using-workflows/caching-dependencies-to-speed-up-workflows#restrictions-for-accessing-a-cache

That is, the caches need to be created in main but I think right now we are creating them on branches so they don't access each other even if they have the same key.

What should we do? 
If I am right about the above I see three options:
1) Stop caching the virtual environment (how long does the virtual environment takes to be created, for core it takes around half a minute but I guess full is much longer, how much?).
2) Reduce the footprint of the virtual environment so each is smaller.
3) Create caches automatically from the main / master branch so the branches have access to them.

I also think that we should implement logic in the testing workflow files so we have either an assertion -stop the tests- or a clean install when there is a cache miss event.

What do you think? I am missing something?

h-mayorquin, 2023-02-14:
Ok, so after discussion with @samuelgarcia on the 2023-02-15 we agreed that we are implementing the solution number 3 above. I already build a solution to have the caches produce in master (see #1321). However, from the [github documentation ](https://docs.github.com/en/actions/using-workflows/caching-dependencies-to-speed-up-workflows#restrictions-for-accessing-a-cache) see the following quote:

> When a cache is created by a workflow run triggered on a pull request, the cache is created for the merge ref (refs/pull/.../merge). Because of this, the cache will have a limited scope and can only be restored by re-runs of the pull request. It cannot be restored by the base branch or other pull requests targeting that base branch.

That is, to finish this issue and test that branches will have reading access to the produce caches I need to have #1321 merged into master. Can we move forward with that?

h-mayorquin, 2023-02-14:
Once #1321 is merged, these are the remaining tasks to improve the CI testing piepeline and close this issue:

- [ ] Verify and test that the branches have read access to the caches produced by this action.
- [ ] Refactor the creation of the virtual environment action. 

The second will allow us to reduce code duplication and standarize the testing pipeline with a common script for virtual env creation. This can be achieved by creating a local action.

 Another possible option is to use the more general and universal mechanism that github actions provides: https://github.com/actions/setup-python#caching-packages-dependencies. However, note that to profit from this mechanism, we will need to use locked-in dependencies (e.g. numpy==1.22) instead of floating ones that we have right now in the toml file.

alejoe91, 2023-02-17:
@h-mayorquin this seems to be fixed to me, right?

h-mayorquin, 2023-02-17:
Yes, the problems is solved. 

There is still an edge case that will cause problems when we change the toml file many times in less than seven weeks (the time that it takes for an unused cache to be erased). I am not sure that this will ever happen and if does once in a while we can solve it by manually erasing the old caches). If this happens very often we should automatize the deletion of old caches but as things stand this is a little bit more involved. 

I am closing this and flagging that situation as someone to keep an eye on.

closed 2023-02-17 by h-mayorquin