SpikeInterface/spikeinterface #1313: Saving/Loading of filtered recording does not return identical traces

pauladkisson, 2023-02-13:
# Current Behavior
1) load NWB file --> recording
2) recording --> bandpass filter --> filtered_recording
3) filtered_recording --> save --> cached recording
4) cached recording --> load --> reloaded_recording
5) filtered_recording != reloaded_recording (small errors in the voltage traces)

# Expected Behavior
loading saved recordings should always return identical traces as the original saved object i.e. saving --> loading --> saving --> ... should not change the data

# Example Code to Reproduce Error
```
from pathlib import Path
import numpy as np
import spikeinterface.extractors as se
import spikeinterface.preprocessing as spre
import spikeinterface.core as sc
import spikeinterface.widgets as sw
import matplotlib.pyplot as plt

# Load the Data
basepath = Path("/Volumes/T7/CatalystNeuro")
filepath = basepath / "TutorialDatasets/sub-npI1_ses-20190413_behavior+ecephys.nwb"
recording = se.read_nwb(filepath)
rec10s = recording.frame_slice(start_frame=0, end_frame=recording.sampling_frequency * 10)

# Preprocess Data
rec_filt = spre.bandpass_filter(rec10s, freq_min=300., freq_max=6000.)

#  Save the Data
job_kwargs = dict(n_jobs=1, chunk_duration='1s', progress_bar=True)
savepath = basepath / "save_load_error"
rec_filt.save(folder=savepath, **job_kwargs)

# Reload the Data
rec_reloaded = sc.load_extractor(savepath)
try:
    assert np.all(rec_reloaded.get_traces() == rec_filt.get_traces()), "Saving/Loading Error"
except AssertionError as e:
    print(e)

# Plot Output
fig, ax = plt.subplots(figsize=(10, 10))
sw.plot_timeseries({"rec_filt":rec_filt, "rec_reloaded":rec_reloaded}, channel_ids=recording.channel_ids[0:1],
                   time_range=[0, 0.001], ax=ax)
fig.show()
```

samuelgarcia, 2023-02-14:
Unfortunatly this is not bug but a feature!
When you save the filtered version the computation is done chunk by chunk for filtering there is a known border effect on every chunk broder. This lead to small diff (<0.1% normally) between computing the entire traces filtered and the chunked traces filtered. We can reduce this error playing with the maring of the filter class.

alejoe91, 2023-02-14:
Exactly! It's

closed 2023-02-14 by alejoe91

pauladkisson, 2023-02-14:
Thank you for the clarification!  I'm still getting used to the chunking/lazy processing, so it surprised me at first, but now it totally makes sense.  I will try to keep this behavior in mind going forward.