SpikeInterface/spikeinterface #1316: import errors after update version 0.98.0.dev0

PaolaGith, 2023-02-14:
Hi, I updated to the latest release, installing from source (https://pypi.org/project/spikeinterface/). Now my spikeinterface version is 0.98.0.dev0. Nevertheless I get import errors and attribute errors when importing the packages, so I am not able to use it anymore.

ImportError: cannot import name 'BaseEvent' from 'spikeinterface' (unknown location)

AttributeError: module 'spikeinterface' has no attribute '__version__'

Has anybody else met these issues?

Thanks a lot!

PaolaGith, 2023-02-15:
Hi! I am running a jupyter notebook script, so as, you suggested, I moved it to another folder and imported. Now I don't get the import error anymore but I get a ModuleNotFoundError.

import spikeinterface as si
import spikeinterface.extractors as se 
import spikeinterface.toolkit as st
import spikeinterface.sorters as ss
import spikeinterface.comparison as sc
import spikeinterface.widgets as sw
from scipy.io import loadmat

ModuleNotFoundError                       Traceback (most recent call last)
Cell In [1], line 3
      1 import spikeinterface as si
      2 import spikeinterface.extractors as se 
----> 3 import spikeinterface.toolkit as st
      4 import spikeinterface.sorters as ss
      5 import spikeinterface.comparison as sc

ModuleNotFoundError: No module named 'spikeinterface.toolkit'

samuelgarcia, 2023-02-15:
Hi Paola.
cool. Yep toolkit was removed 2 releases before a kept for while now now this is over.
All function have been moved at the correct place (preprocessing, postprocessing, curation or qualitymetrics)
remove_artifacts and common_reference are in preprocessing now.

h-mayorquin, 2023-02-16:
I guess we can close this issue @PaolaGith ?

PaolaGith, 2023-02-16:
Sure!

closed 2023-02-16 by PaolaGith