SpikeInterface/spikeinterface #1322: Troubleshooting SpikeInterface spike sorting with TDT recordings

bradlytstone, 2023-02-15:
I have been attempting to employ SpikeInterface to process (pre-process and spike sort) data that my lab has collected using the TuckerDavis (TDT) system. Unfortunately, however, after trying it several different ways, and many combinations of parameters (across many sorters: Kilosort, spykcircus2, trideclous), I am unable to resolve units (nor waveforms) that look right. One major issue appears when I go to verify the "sorted units" and see that the waveforms are inverted for nearly all of them (see attached). Online forums suggested that the probe and/or channel_groups were not properly set, some forums mentioned that the thresholds are off. I have exhausted nearly all attempts to get this to work thus I have turned here for assistance/feedback on what I am doing wrong? I can also provide sample datasets as to see if others replicate (or fix) the issues I mentioned (they are too large to place here).

For clarity in probe specifics,  I performed the following within my pipeline.

# Design the probe based on experiment parameters (we used the NeuroNexus A1x32-Poly2)
probe = generate_multi_columns_probe(num_columns=2,
                                     xpitch=43.3, ypitch=50,
                                     num_contact_per_column=[16, 16],
                                     y_shift_per_column=[0, 25],
                                     contact_shapes='circle', contact_shape_params={'radius': 7.5})

# Provide channels that are remapped
device_channel_indices=[7,5,3,1,30,28,25,27,6,29,4,31,2,24,0,26,9,11,\
                        13,15,16,18,23,21,8,19,10,17,12,22,14,20]

# Set probe within reader object
reader = reader.set_probe(probe)

[image]
[image]
[image]
[image]

alejoe91, 2023-02-16:
Hi @bradlytstone

The "inverted" waveforms could be due to a number of reasons. First, what do you mean with inverted exactly? The waveforms should have their peak in the negative direction. Looking at plot 3, that seems to be the case (and a good unit, but rather small).
Your first, second, and last plot in my opinion are noisy units, because signals are very similar across many channels. 

When you plot the raw data, do you clearly see many spikes? You can try with the `sw.plot_timeseries()` function (see [docs](https://spikeinterface.readthedocs.io/en/0.97.0/modules/widgets.html) and [function](https://spikeinterface.readthedocs.io/en/0.97.0/api.html#spikeinterface.widgets.plot_timeseries))

bradlytstone, 2023-02-16:
Hi @alejoe91 

Thanks for taking a look at this. Jumping to your last question, yes. I have looked at the raw data....and upon visual inspection (and zooming into areas where putative spikes are) I do see spikes. Albeit sparse, given our region and duration of recording (we are still testing things). As for the first question, what I mean by inverted is that is appears that I consistently get waveforms that are overlain in opposing directions -- with respect to their peak. If you look at the example below, I have outlined the two opposing waveforms (green and black). It almost looks as if they are mirror images of one another, or that the sorting algorithm did somethin regarding the polarity of the signal. Have you seen this before, and if so, do you know how to fix this? Really....if we can resolve this issue, then I can move forward with the pipeline and begin verifying the 'sorted units.'

[image]

alejoe91, 2023-02-16:
So in general very often this positive peaks that seems to mirror the negative ones (but most of the times are smaller) could be due to dendritic signals. This is a well known phenomenon, as dendrites kind of balance the currents from the perisomatic region. Nevertheless, in this specific example I would say that the waveforms are too similar across channels (both the negative and positive ones), so this is probably some very coherent noise.

Honestly, I don't think that the spike sorting is doing anything funky. Note that the spike sorter does not change on the traces! And waveforms are extracted afterwards only using the spike sorting output and the recording. So if this is what you get out of the spike sorter, then it means that it is in the recording (although it might be hard to spot from a visual inspection).

bradlytstone, 2023-02-16:
Thank you for providing that information. That draws me to another question: is it advised to (lets say for the example 'green/black' waveform I sent) to remove/exclude any/all of these "positive peak" waveforms from downstream analyses?

I, no doubt, believe our recordings are noisy -- this is somethin we are currently working on to fix.

The recording that those examples came from was ~10-15minutes. I also sent those recording files, along with another set, via a box link (in email) if its easier to see the data for yourself. Thank you again!

alejoe91, 2023-02-16:
I would advise to remove very noisy channels (we have [automated functions for it](https://spikeinterface.readthedocs.io/en/latest/modules/preprocessing.html#detect-bad-channels-interpolate-bad-channels)), but not removing "manually" channels after spike sorting. In this case, I would just remove the unit from downstream analysis (if it's noisy, it will be on all channels!).

My advice is: if the timeseries look correct (noise levels, channel mapping, etc.) the spike sorting will be fine once the recordings are optimized ;) So I would spend time trying to make recordings better now and think about analysis later (again, good recordings --> better spike sorting).

bradlytstone, 2023-02-16:
Okay, thank you! I will work on this a bit more!

bradlytstone, 2023-02-16:
In thinking about this further, I realize my question may have been misunderstood. I am not manually removing channels post-spike sorting. What I am asking is whether, post-spike sorting, in cases where I see a single 'sorted_unit' have both negative and positive waveforms overlain (as in that green/black example), it is valid (and/or I should) to detect instances of the waveforms with the peak in the positive direction, and remove those....thus leaving timepoints where the peak is in the negative direction as actual synaptic spikes?

alejoe91, 2023-02-16:
Closing for now! Feel free to open another issue later

alejoe91, 2023-02-16:
I think you should just remove the entire unit, since it's probably noisy or corrupted. If only a few channels have a positive deflection, that is ok! But in the examples you shared it seems noisy to me

closed 2023-02-22 by alejoe91