SpikeInterface/spikeinterface #1323: Issue while running mountainsort4

Hugues1273, 2023-02-15:
Dear all, 

I am trying to use Mountain Sort 4 sorter through SpikeInterface (of course) and I get this error message:

> Traceback (most recent call last):
> 
>   File "E:\Python\SpikeInterface2.py", line 95, in <module>
>     sorting = ss.run_sorter("mountainsort4", recording_cmr, output_folder = "C:/Users/lamot/Documents/TestMS6", verbose = True, docker_image = True, **ms4_params)
> 
>   File "C:\Users\lamot\Anaconda3\lib\site-packages\spikeinterface\sorters\runsorter.py", line 137, in run_sorter
>     return run_sorter_container(
> 
>   File "C:\Users\lamot\Anaconda3\lib\site-packages\spikeinterface\sorters\runsorter.py", line 578, in run_sorter_container
>     raise SpikeSortingError(
> 
> SpikeSortingError: Spike sorting in docker failed with the following error:
> b'/usr/local/lib/python3.8/site-packages/spikeinterface/preprocessing/whiten.py:53: RuntimeWarning: divide by zero encountered in divide\n  W = (U @ np.diag(1 / np.sqrt(S))) @ Ut\nwhitening\nMountainsort4 use the OLD spikeextractors mapped with NewToOldRecording\nUsing temporary directory None\nUsing 1 workers.\nError running mountainsort4\nTraceback (most recent call last):\n  File "/usr/local/lib/python3.8/site-packages/spikeinterface/sorters/basesorter.py", line 226, in run_from_folder\n    SorterClass._run_from_folder(sorter_output_folder, sorter_params, verbose)\n  File "/usr/local/lib/python3.8/site-packages/spikeinterface/sorters/mountainsort4/mountainsort4.py", line 133, in _run_from_folder\n    old_api_sorting = mountainsort4.mountainsort4(**ms4_params)\n  File "/usr/local/lib/python3.8/site-packages/mountainsort4/mountainsort4.py", line 22, in mountainsort4\n    geom = _get_geom_from_recording(recording)\n  File "/usr/local/lib/python3.8/site-packages/mountainsort4/mountainsort4.py", line 62, in _get_geom_from_recording\n    location0 = recording.get_channel_property(channel_ids[0], \'location\')\n  File "/usr/local/lib/python3.8/site-packages/spikeinterface/core/old_api_utils.py", line 46, in get_channel_property\n    v = values[ind[0]]\nTypeError: \'NoneType\' object is not subscriptable\n\nTraceback (most recent call last):\n  File "/Users/lamot/Documents/in_container_sorter_script.py", line 17, in <module>\n    sorting = run_sorter_local(\n  File "/usr/local/lib/python3.8/site-packages/spikeinterface/sorters/runsorter.py", line 163, in run_sorter_local\n    SorterClass.run_from_folder(output_folder, raise_error, verbose)\n  File "/usr/local/lib/python3.8/site-packages/spikeinterface/sorters/basesorter.py", line 262, in run_from_folder\n    raise SpikeSortingError(\nspikeinterface.sorters.utils.misc.SpikeSortingError: Spike sorting failed. You can inspect the runtime trace in /Users/lamot/Documents/TestMS6/spikeinterface_log.json\n'
> 
> 

And here is the content of the spikeinterfact_log.json:

> {
>     "sorter_name": "mountainsort4",
>     "sorter_version": "1.0.3",
>     "datetime": "2023-02-15T16:05:10.332341",
>     "runtime_trace": [],
>     "error": true,
>     "error_trace": "Traceback (most recent call last):\n  File \"/usr/local/lib/python3.8/site-packages/spikeinterface/sorters/basesorter.py\", line 226, in run_from_folder\n    SorterClass._run_from_folder(sorter_output_folder, sorter_params, verbose)\n  File \"/usr/local/lib/python3.8/site-packages/spikeinterface/sorters/mountainsort4/mountainsort4.py\", line 133, in _run_from_folder\n    old_api_sorting = mountainsort4.mountainsort4(**ms4_params)\n  File \"/usr/local/lib/python3.8/site-packages/mountainsort4/mountainsort4.py\", line 22, in mountainsort4\n    geom = _get_geom_from_recording(recording)\n  File \"/usr/local/lib/python3.8/site-packages/mountainsort4/mountainsort4.py\", line 62, in _get_geom_from_recording\n    location0 = recording.get_channel_property(channel_ids[0], 'location')\n  File \"/usr/local/lib/python3.8/site-packages/spikeinterface/core/old_api_utils.py\", line 46, in get_channel_property\n    v = values[ind[0]]\nTypeError: 'NoneType' object is not subscriptable\n",
>     "run_time": null
> }

I tried to modify the detection parameter putting it as low as 1.01 but it did not work, 

Then, as I saw that the runtime_trace was empty in the json file, therefore I tried to put the folder for both recording.save() and ss.run_sorter(output_folder = "") but in then an error message (permission denied) is raising...

do someone has an idea to fix it?

Thank you so much!

alejoe91, 2023-02-16:
Hi @Hugues1273 

Can you paste your entire code? I'll try to reproduce the error on my side

Hugues1273, 2023-02-16:
Thank you so much for your reactivity, so kind!
So I paste my code:

```
import spikeinterface
import spikeinterface.extractors as se
import spikeinterface.sorters as ss
import spikeinterface.comparison as sc
import spikeinterface.widgets as sw

import matplotlib.pyplot as plt
import numpy as np
import os

import docker

num_channels = 2 ## I know it is low, it is intracranial recording in humans
sampling_frequency = 48000

subjectpath = "C:/Users/lamot/Documents/Test/CDEV_GPI_12_sec_-5.83.raw"

recording = se.BinaryRecordingExtractor(subjectpath, dtype = 'float32', num_chan = num_channels, sampling_frequency = sampling_frequency)

print(recording)

channel_ids = recording.get_channel_ids()
fs = recording.get_sampling_frequency()
dtypeID = recording.get_dtype()
num_chan = recording.get_num_channels()

print(f"Channel IDs: {channel_ids}")
print(f"Sampling frequency: {fs}")
print(f"Binary type: {dtypeID}")
print(f"Number of channels: {num_chan}")

trace_snippet = recording.get_traces(start_frame = int(fs * 0))
print("Traces shape:", trace_snippet.shape)

total_duration = trace_snippet.shape[0]/fs

w_ts = sw.plot_timeseries(recording, time_range = (0, total_duration))
plt.show()

import probeinterface as pi
import probeinterface.plotting as piplt

n = 2
Positions = np.zeros((n, 2))
Positions[0] = 0, 50
Positions[1] = 0, 150
recording_prb = pi.Probe(ndim = 2, si_units = "um")
recording_prb.set_contacts(positions = Positions, shapes = "circle", shape_params = {"radius": 20})
print(recording_prb)
polygone = [(-20, -30), (0, -110), (20, -30), (20, 190), (-20, 190)] ## à modifier si je veux modifier la taille de l'électrode
## en particulier il faudra que j'ai la largeur pour l'adapter, ainsi que le radius (sachant que pour spyking circus on l'a
## considéré à 100 et qu'ici j'ai mis 20)
recording_prb.set_planar_contour(polygone)
piplt.plot_probe(recording_prb)

print(recording.get_binary_description())
print(recording.get_total_duration())

from spikeinterface.preprocessing import bandpass_filter, common_reference

recording_f = bandpass_filter(recording, freq_min = 300, freq_max = 6000)
recording_cmr = common_reference(recording_f, operator = "median", reference = "global")

w = sw.plot_timeseries(recording_cmr, color = "red")

print(ss.installed_sorters())

recording = recording.save(folder = "C:/Users/lamot/Documents/TestMS7")
ms4_params = ss.get_default_sorter_params("mountainsort4")
ms4_params["detect_threshold"] = 1.01
ms4_params["filter"] = False
sorting = ss.run_sorter("mountainsort4", recording_cmr, output_folder = "C:/Users/lamot/Documents/TestMS7/Results", verbose = True, docker_image = True, **ms4_params)
print("sorting", len(sorting.get_unit.ids()), "units")
```

Sorry for my beginner level code :D

Thank you so much again!

Hugues

alejoe91, 2023-02-16:
Hey @Hugues1273 

No worries for the beginner's level, we have all been there ;)

I tracked down the problem:
You are creating a probe, but you are not setting it! So mountainsort tries to retrieve channel locations, but fails..

Add these two lines before the preprocessing:
```
# this line "wires" the probe cpntacts to the acquired channels
recording_prb.set_device_channel_indices([0, 1])
# this line actually attaches the probe object to the recording
recording = recording.set_probe(recording_prb)
```

After this, MS4 should run smoothly!!!

Let me know if this fixes it on your side

PS: you have a small typo in the last line: `sorting.get_unit.ids()` --> `sorting.get_unit_ids()`

Hugues1273, 2023-02-16:
Ok, I had some trouble with anaconda (spyder failed to launch (???)), so I am reinstalling it and I tell if the code correction is working!
Thank you so much again to take time for me!

Hugues

Hugues1273, 2023-02-16:
it wooooooooorks!!!!!
Thank you so much, I am so happppyyyyyyyy!!!

Thank you!

closed 2023-02-17 by alejoe91